arendst/Tasmota #3134: Set button action on T1

nevembalint, 2018-07-05:
Hi, I was wondering if it's possible to set what each button does on a T1. I want to operate a roller shutter with it. I would set the leftmost button to set relay 1 ON and set relay 2 ON and, the rightmost button to set relay 1 to ON and relay 2 OFF and the middle button to set relay 1 OFF. So I'd only use two relays, one which is cutting power to the other (when I press the "stop" button) and the other relay would control the roller shutter motor (ON: give power to MOTOR UP and OFF: give power to MOTOR DOWN on the roller shutter). Does this make sense?

Frogmore42, 2018-07-05:
Simple answer: no
Assuming you have a three position Touch switch, the relays are all independent, so without rewiring it, you can't do what you want.

ascillato2, 2018-07-16:
Hi

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2

nevembalint, 2018-07-31:
Yeah, i know i have to rewire it, that's not a problem. But when i do, will I be able to do the button settings?