arendst/Tasmota #3135: how to configre sonoff basic with RCWL0516 ?

aihysp, 2018-07-05:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

using onoff basic with RCWL0516 rader

when i connect the sensor all is great but instead of sending MQTT command for topic sensor 
the relay itself turns on and off according to the rader 
i guess there is an option to cancel it , and connect the sensor to MQTT topic 
in the end i want to be able to see on node red status for that sensor

found this issue but nothing useful there, too many commands can someone point me?
https://github.com/arendst/Sonoff-Tasmota/issues/247

thanks!

reloxx13, 2018-07-05:
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic#switchtopic

aihysp, 2018-07-05:
still not able to config

can anyone guide me how to set PIR or 0516 connected to the sonoff
maybe i need some resistor

aihysp, 2018-07-06:
ok so sensor is connected to gpio 14
when there is movement the realy turns of and on, so it looks like it working but not as i want it

on logs i am not able to see a new topic for gpio 14
only that relay turns on and off

Frogmore42, 2018-07-06:
You don't have it configured correctly.  How did you configure it?  Please provide the log output of Status 0 as is always required to provide adequate help.

aihysp, 2018-07-06:
i hope this is what you mean 

00:20 MQT: tele/PIR1/STATE = {"Time":"2018-07-06T06:00:20","Uptime":"0T00:00:18","Vcc":3.176,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Ofir-grid","RSSI":36,"APMac":"C8:BE:19:09:BD:29"}}

i configured gpio14 as a relay

i have played around allot with switch mode and topic so i am not sure what is the config there

Frogmore42, 2018-07-06:
In the console type:
Status 0

It will print a bunch of stuff that will include how you have it configured.

Reread the section in the wiki on switch mode and topic.  They are pretty well described there.  Switch Mode determines if an MQTT message is sent when a switch is triggered or if the relay is controlled directly.  Switch Topic is the non-default message to send.  You have to change both if you want a custom message.  You also need to keep good notes about what you tried and what behavior you see in order to really understand how this all works.

aihysp, 2018-07-06:
wow thanks

06:47:44 CMD: status 0
06:47:44 MQT: stat/PIR1/STATUS = {"Status":{"Module":1,"FriendlyName":["outsideLight","Sonoff2"],"Topic":"PIR1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
06:47:44 MQT: stat/PIR1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:21","StartupUTC":"2018-07-06T05:47:23","Sleep":0,"BootCount":70,"SaveCount":366,"SaveAddress":"F6000"}}
06:47:44 MQT: stat/PIR1/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
06:47:44 MQT: stat/PIR1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Ofir-grid",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
06:47:44 MQT: stat/PIR1/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
06:47:44 MQT: stat/PIR1/STATUS5 = {"StatusNET":{"Hostname":"PIR1-0828","IPAddress":"10.0.0.34","Gateway":"10.0.0.138","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.138","Mac":"DC:4F:22:2A:03:3C","Webserver":2,"WifiConfig":2}}
06:47:44 MQT: stat/PIR1/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2A033C","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
06:47:44 MQT: stat/PIR1/STATUS7 = {"StatusTIM":{"UTC":"Fri Jul 06 05:47:44 2018","Local":"Fri Jul 06 06:47:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:54","Sunset":"20:55"}}
06:47:44 MQT: stat/PIR1/STATUS10 = {"StatusSNS":{"Time":"2018-07-06T06:47:44"}}
06:47:44 MQT: stat/PIR1/STATUS11 = {"StatusSTS":{"Time":"2018-07-06T06:47:44","Uptime":"0T00:00:21","Vcc":3.180,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Ofir-grid","RSSI":32,"APMac":"C8:BE:19:09:BD:29"}}}

Frogmore42, 2018-07-06:
Now, what are you using for switchmode and switchtopic?
If you type the command without parameters it will tell you its current setting.

aihysp, 2018-07-06:
switchmode 1

Frogmore42, 2018-07-06:
Try setting switchtopic to 1
It sounds like it didn't pick up the change to switch topic if it is still controlling the relay.

Also please cut and paste the command and reply from the console so there are no typos.

aihysp, 2018-07-06:
stat/PIR1/RESULT = {"SwitchMode1":1}
stat/PIR1/RESULT = {"SwitchTopic":"PIR1"}

maybe this rader sensor needs more juice? i am powring it from the sonoff 3.2 rail

Frogmore42, 2018-07-06:
When dealing with unknown things, one must be very methodical and keep excellent track of what was tried and the results.  Not doing that generally leads to confusion.  You have been doing about half of what I requested.  While having a switch send out information via MQTT is not hard and hooking up an rcwl sensor isn't either.  If you have never done this before and don't have a working knowledge of electronics and computer science/software, it can be VERY confusing.  I know many people who would never be able to figure it out.  Since you were able to get something to work somewhat, I assume you are capable of getting this to work.  But, it will take a lot of patience and a VERY methodical approach.

Currently, it appears the sensor is not working.  It might be broken or it might be misconfigured.  In order to tell, remove the sensor and replace it with a switch.  With the switch in place, you can be more certain of when there should be something in the log.  Now experiment with the setup.  Set switchmode to 1 and switchtopic to 1.  This should get you to seeing MQTT messages every time the switch is changed.  If that doesn't work, try starting over.  Completely erase the ESP and then reflash and reconfigure it.  Doing this will eliminate the confusion that a possibly defective RWL sensor could cause.

aihysp, 2018-07-06:
thank you I will try this, lucky me I ordered two 
and yes you guessed right, not a lot of knowledge but i am trying :)

maybe i will go back to ESPEasy, there i can find the logic 
Tasmota is great but too many commands ,  and not allot of GUI where i shine haha

Frogmore42, 2018-07-06:
If ESPEasy does what you need, great.  I looked at it and found it confusing to do anything beyond very basic stuff.  I also found Tasmota to be very confusing when I first started using it.  I expect that had I stuck with ESPEasy I probably would have found it to be easy eventually.  There are many concepts that you need to understand in order to fully use Tasmota, but you can still get some benefit with minimal understanding.

Because Tasmota is used by a LOT of people with different needs, it provides a lot of choices.  Any time there is a choice, there is a possibility of confusion.  For example, when starting out, how do you install the software?  The wiki lists several ways, most of which lead people down the path of compiling the code.  This is not easy, if one has not done it before.  Even if one has, there have been issues with underlying frameworks upon which Tasmota depends, that make it more complicated.  I remember when I first started, I just wanted to download an executable.  I gave up trying to figure out how to do that.  I chose the PlatformIO path instead.  That was confusing, even though I am a trained professional 😀.  I had not used it before and it operates differently than other tools I had used.  I was able to figure it out, but the cfg_holder usage definitely confused me until I took the time to understand how it works.  It isn't the way I would have designed it, but I can see why Theo chose that way.  Bottom line, if you are willing to take the time to learn things, you will find Tasmota to be good.  If not, you might be better served by one of the turn-key solutions like SmartThings.  I have one, and like some things about it, but find it to be too constrainting and the requirement for cloud connectivity for just about anything to work is too much for me to use it for most things.

The good news is there is choice.  Make the one(s) that work for you.

aihysp, 2018-07-07:
i think i will close the issue, as i am not sure what to ask...
i will do some more testing, thank you so much for all the help

closed 2018-07-07 by aihysp