arendst/Tasmota #3137: Is it possible to use sonoff to control a blind?

yonaesp, 2018-07-05:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ionciubotaru, 2018-07-06:
https://github.com/arendst/Sonoff-Tasmota/issues/2212

yonaesp, 2018-07-06:
it really does not help me, since I know that with sonoff 4ch r2 I can do it, there does not say anything of sonoff dual

ionciubotaru, 2018-07-06:
Two relays are enough to control blinds, one form up and another for down.
To determine the exact position of your blinds use reed sensor and magnets.
Tasmota just read it, your HA must do the maths.

yonaesp, 2018-07-06:
I want to do everything only with the sonoff, no external server, much less place magnets and sensors.

If it is not possible with the sonoff dual, I will use the sonoff ch4 r2

Leseratte10, 2018-07-06:
What makes you think that this would be possible with the 4CH R2 device? Something like blinds position, or moving them to a specific position, is handled by your home automation system like iobroker or HomeAssistant. Not by Tasmota.

HGBAXXTER, 2018-07-06:
If the blinds are not only controlled by home automation but also via buttons directly connected to the Sonoff, the home automation would not be aware about this. In my opinion, it would be ideal to do the maths on the sonoff, just like it is done e.g. in Homematic shutter actuator. MQTT commands like “set blind 40%“ only based on timing and not relying on magnets or sensors would be a great feature.

I wonder that there seems to be so little interest to implement such a feature. I assume that there are many enthusiastic Tasmota users who are exactly waiting for that (including myself).

Frogmore42, 2018-07-06:
Take a look at the referenced issue above.  Then take a look at stefanbode's repo.  I believe he has already done what you want.  Making it work on a Dual should be possible with some reconfiguration, but I guess he would be able to help you there.

While there are some people who have rolling shutters, I have never even seen one in person and I have to guess they are much less common than light bulbs 😀

HGBAXXTER, 2018-07-06:
I was just looking at the referenced issue but unfortunately I seem to have bricked my Dual during flashing :-( Will continue during the weekend. 

Probably, I was not precise enough about the interest that I meant that people may have. I am sure, that for "real" shades, shutters or blinds, most people would tend to a professional solution (because of "real" hardware interlock, or because it is probably not so simple to integrate a Dual or 4ch into a concrete wall next to a window...). What I am thinking of are more "nice to have" things. In my example, I am trying to automate an electric arm chair. I think it is a pretty cool thing to tell Alexa to set my sofa to 50%. Spending 10 € on that is worth the fun, but investing in e.g. a Homematic based solution might be a little over the top. 

I am monitoring this topic quite a while, and every couple of days a new issue on this topic is opened. So there seems to be some interest. However, most of them are closed without further action (maybe the interest is not significant enough). As there seem to be working forks around, I would personally be very happy, if sometime one of these implementations would be merged into the orignal branch.

Frogmore42, 2018-07-07:
They are closed because Tasmota already has basic control capability for rolling shutters.  You want something more.  But, implementing that something more reliably without additional hardware is probably not possible.  Implementing what you seem to want, in a way that is good enough for some people, has also already been done, by people who have taken Tasmota and modified it and shared it.

Having seen the trouble people have with the most basic of tasks necessary to get Tasmota running, I am not convinced it would be good to add something more that is likely to be confusing.  But, luckily for your desires, it is not I who makes the decision about what gets added and what doesn't.

yonaesp, 2018-07-08:
@HGBAXXTER I bricked the device through an update via wifi.

Luckily I could unbrike it, now before installing a firmware I should install minimal first.

**What I ask maybe is something like this:**
Dual Sonoff has 2 buttons (up and down) and the blinds take 10 seconds to fully upload.
Add virtual votes in this way:
Up
Up 1 (with 2 seconds timer) (blinds at 20%)
Up 2 (with 5 seconds timer) (blinds at 50%)
Up 3 (with 8 seconds timer) (blinds at 80%)
Down
Down 1 (with 2 seconds timer) (blinds at 80%)
Down 2 (with 5 seconds timer) (blinds at 50%)
Down 3 (with 8 seconds timer) (blinds at 20%)

in this way I could say "turn on down 1" to fix the blind at 80% when it is open

joba-1, 2018-07-08:
As far as I know, the buttons of a 2ch SonOff control one line of each output independently.
A good solution for shutters must prevent both lines being on at the same time with hardware (i.e. wiring), or else you risk frying the motors (or much worse...).
How can this be done with 2ch SonOffs? I think you need at least 2 relays with NC and NO outputs accessible.

yonaesp, 2018-07-09:
@joba-1 Yes, with tasmota you can enable interlock mode with the command: SetOption14 1
I have also found a firmware based on tasmota designed for blinds that when you try to turn off or turn on any button, if one of the two is on it turns off, in addition to setting the opening time.

Despite this, I would like to add virtual buttons with different pulse times to put the shutter in different positions

@Leseratte10 I was wrong, I can use dual sonoff in the same way that they sonoff 4ch.

The only difference is that 4ch has interlock mode by hardware, and dual by software, but this do not suppose any problem.

joba-1, 2018-07-10:
@yonaesp: but that is exactly what I meant: if you rely only on software to not switch both directions on and your motor has no internal protection against this, then murphy will kick in: what can fail, eventually will fail. And then potentially very bad things can happen (like burning your house, ...). First thing you learn when you do this professionaly: never, ever rely on software for safety. 
I think it is possible to do safely with a nodemcu and 2 optocoupled relays, but not witha 2ch sonoff where you cannot acces the NC relay pins.

yonaesp, 2018-07-10:
@joba-1 I have tried to restart the sonoff several times and it has not lost the configuration, I think it is something very difficult to happen, I have known cases of sonoff that lose the configuration when restarting, but they are problems of some versions in particular.

Any device can fail, a car also uses software, a lot has not triggered the brake sensor and have ended up running over people (of the new cars).

If it is true that it can be made safer with a relay, but I believe that I will take the risk, always doing the tests with light bulbs of course.

stefanbode, 2018-07-16:
The Interlock works very well, but anyhow I agree that better safe than sorry. A design on the mechanical side is much better. One typical configuration to overcome the issue is to use one relay for switching ON/OFF the power and the second relay as a toggle getting the power from the first relay to the common pin and then toggle it left or right, or better up and down. The good thing is you do not need to change anything on the existing hardware. In my branch of TASMOTA I have currently two kinds of shutters implemented. the normal ones and the ones that work with pulses. There are also safe to use. It should be not a big deal to implement the third mode like explained above. The Relay modules you can buy everywhere have these toggle relays with 3 pins. At least one of the Sonos, too.

stefanbode, 2018-07-16:
Yep, no big deal. Current version supports the described above short-circuit safe version

ascillato2, 2018-08-25:
Just for reference. Similar issues:
#2212
#3176

ascillato, 2018-09-04:
Let's put similar issues into one: https://github.com/arendst/Sonoff-Tasmota/issues/288

closed 2018-09-04 by ascillato2