arendst/Tasmota #3138: Nueva propuesta de logotipo para Sonoff-Tasmota AND Sonoff-tasmota_KNX

orlant, 2018-07-06:
For me, it is a pleasure to introduce you to the idea arising from the union of the initials of Sonoff Tasmota.

S + T + network design. ... this logo is very easy to recognize. even if we remove the letters you can use it with an ideal LogoMark for that project

[image]
[image]

application of the logo in the firmware
[image]
[image]

I hope you like my idea, any modification of colors and font you want I am ordering ... there are both versions.

arendst, 2018-07-06:
Nice idea.

To add the logo to the webpage how many bytes would be have to be used keeping in mind that any bytes used for the logo cannot be used for functionality ;-)

orlant, 2018-07-06:
Does it refer to the number of bytes that the design has? Is there a recommendation or modification that I could make for your use ??? possible solution is that I can minimize the number of bytes of the logo ...

arendst, 2018-07-06:
I hope the logo would not use as much as up to 60 bytes ….

orlant, 2018-07-06:
To complete this task by personal recommendation we can leave the LOGO / ICON ... which does not contain the letters of the project, for the use of its functionality it helps to avoid the number of bytes ... anyway I will reduce to 60 bytes all the designs made seems ... I'll show you

orlant, 2018-07-06:
I'm sorry sir theo 60 bytes is very little and it will not even be seen how small it would be, if I could use at least 1kb I could do something for you !!!

ascillato, 2018-07-06:
Hi,

@orlant 

Sonoff Tasmota is mainly a MQTT device with an extra of KNX between other functions (HTTP, SERIAL, TIMERS, etc), so I think it is not needed the KNX name. Also, I have just realized that It is going to be a trademark problem using a registered brand name as KNX and also SONOFF. Both names are trademark protected and can not be used.

I think that you should use just TASMOTA and a Icon.

_(Also, please edit the title of this issue to english)_

@arendst 

If you want add a logo, one option that don't use flash memory space could be just have the logo at your repository and be linked from the Tasmota webpage (if the device have internet, the logo will be displayed, if not, just a name will be shown)

orlant, 2018-07-06:
ok I am willing to make the changes that are important, with pleasure but it would be a shame because Mr. theo requires me that the logo has a maximum size of 60 bytes, it is very low resolution and the design is lost ...
If the proposal that you have given Mr. Theo takes it into account, it would be great because you can see the result obtained added to your repository. that would be the first benefit and second benefit will be visible on the website. if you think it is necessary I can make the modifications.
"Let's hope Mr. Theo says" ..

(@ascillato I've edited the title from Spanish to English three times and it's still the same, I'll try again)

ascillato, 2018-07-06:
Hi,

@orlant 

You write first in my repository at https://github.com/ascillato/Sonoff-Tasmota_KNX/issues/151 and I was trying to explain to you that there is not a separate software with KNX. There is just this Tasmota Repository. The other is the same but I use that for tests. The original is Theo's.
That's why, a logo should be without the KNX word, also due to it is registered to knx.org

ascillato, 2018-07-06:
Also a better resolution logo can be used in this repository.

orlant, 2018-07-06:
perfect and I am much more understood ... I will make all the necessary corrections and I will work based on the LOGOTasmota Root Project ... if it is necessary in the future it makes some edition I commit myself with you to be here!

orlant, 2018-07-06:
I would like to show you the modifications I have made for you! along with a variation of the ST logo model in black and white version   

[image]

wongnam, 2018-07-07:
please be aware of license right, you should not put the "Sonoff" before the "TASMOTA" word. because it's a brand name off Itead products.
i suggest only use TASMOTA to create a logo is enough the meaning.

orlant, 2018-07-07:
Thanks a lot for the suggestion. well there and I made the correct modification.

wongnam, 2018-07-07:
as my idea for the modern logos world, a logo is always simple and easy to identify is the best.

btw, i very like the only text in your design to use as TASMOTA logo :)
[image]

pascalsaul, 2018-07-07:
Best solution is to use Base64 encoding so you don't need internet en you can embed it into the HTML.

See: https://varvy.com/pagespeed/base64-images.html

roguestreak, 2018-07-16:
I reckon the kerning on the left of the "A" needs reducing in both cases, tucking it under the "T" slightly to reduce the apparent gap.  I'd also slightly increase the kerning in general for the "SMO" as to me they are a bit squished.

closed 2018-07-16 by orlant