arendst/Tasmota #3139: 4CH Pro WiFi LED

meingraham, 2018-07-06:
TASMOTA 6.0.0c
The Wi-Fi LED only comes on when one of the other channel LEDs is on. If all four channels are off, the Wi-Fi LED is off.

RuhanSA079, 2018-07-06:
Hi, @meingraham 

The default LED behaviour is defined here: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main under LedPower and LedState.

For example, you want to set the LED to blink when a MQTT message is sent, send the following command to the device, assuming the default mqtt topic: "cmnd/<sonoffDeviceTopic>/LedState" with the payload of "7". (Power state and MQTT messages)

meingraham, 2018-07-06:
OK, maybe I misread the documentation... but this seems like odd behavior regardless of the setting.

If **all** four channels are off, all five LEDs are off. If I turn on **any** channel, that channel's LED comes on as well as the blue LED. The channel LEDs "follow" the channel state: power on - LED on, power off - LED off. Blue LED is: any channel on - LED on. ALL channels off - LED off.

Frogmore42, 2018-07-06:
The channel LEDs are hardwired to the relays, so they come on when the relay comes on.  The other LED has configurable behavior.  I actually like the default, since I don't want a flashing light.  The other possible default, always off, doesn't seem right either.  But other people like other things, hence the command to change the behavior.

meingraham, 2018-07-06:
The blue LED is on solid when it's on, i.e., not blinking.

Ideally, what I'd like is for it to blink slowly when the device's Wi-Fi is not connected to the LAN and stay off when it does have a connection. I'd settle for having that blue LED on solid when there is a Wi-Fi connection. I didn't see in the documentation where an LED could display the state of the Wi-Fi connection. The channel LEDs will indicate power status so there's no need for the blue LED to be a power indicator.

On an LED associated with the channel, the default is to turn on the LED when the relay is latched.  The behavior could be that when the device has not established a Wi-Fi connection, then the LED slow blinks.

By the time one of the channel LEDs is on, I already know that the 4CH has power so the blue LED is redundant. I might as well set it to be always off because it's only on when one of the other LEDs is on.

meingraham, 2018-07-07:
Further... more reason as to why having the blue "power" LED "follow" the state of the channel LEDs is odd.

For the 4CH Pro:
LedPower & LedState commands appear to apply to the blue LED only. The channel LEDs follow the latching of the relay and cannot be disabled.

FYI - I had originally set up my 4CH with the Sonoff factory image on eWeLink. In that case, toggling the channel relays did not affect the blue "Wi-Fi state" LED.

Mike

Frogmore42, 2018-07-07:
For the 4CH Pro, you are right. Having two LEDs that are basically the same, is probably not very helpful.  Tasmota does support more generic hardware, where the definitions for LEDs make more sense.  I am not sure how hard it would be to extend the behavior to include the option to show not connected.  If you have the ability to make the code change, it would be a whole lot easier to convince Theo that it needs to be added as an option.  I am pretty sure that is how many options have been added.

meingraham, 2018-07-07:
You don't want me anywhere near the code ;-) My programming skills are rusty, **at best**. Then I'd have to actually figure out how to program how to retrieve the Wi-Fi status and then how to program the LED. For now, I'll settle into the role of idea man ;-) I can post a feature request and hope that someone take on the challenge!

Mike

closed 2018-07-11 by meingraham