arendst/Tasmota #3140: Override loglevel per command

i3laze, 2018-07-06:
Weblog of my device is overrun by STATE requests every 5 minutes.
I didn’t find how to increase weblog buffer (aprox. 35 strings for now)
This way I can’t view what happened 6 hours ago.
I also find deploying external syslog too excessive.

As I mentioned https://github.com/arendst/Sonoff-Tasmota/issues/2843#issuecomment-393448953,
It’s gonna be nice for the user to be able to move some commands between log levels.
Like STATE > lvl 3, SRC > lvl 2

Jason2866, 2018-07-06:
Due to limited Memory of the ESP there is simply "no room" for storing more...
If you need things like that you have to do it extern.
Maybe @arendst finds your idea nice and it is possible to realize LOG Level defining

arendst, 2018-07-06:
As loglevels are incremental there is always the problem of having too much info at some time. The current log levels are fine and more will just open up a big can of worms where some want it on level 10 and others on level 12...

As you already noticed the external syslog server is the way to go. If you have a Linux server around logging is just some small config on the server to make it available for all your devices.

i3laze, 2018-07-06:
Thanks. Seems external syslog with filtering is a way to go.

closed 2018-07-06 by i3laze