arendst/Tasmota #3144: with 6.1.0 flash sonoff stays in minimal firmware

HFeenstra, 2018-07-06:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

<img width="449" alt="schermafdruk 2018-07-06 21 21 36" src="https://user-images.githubusercontent.com/25987376/42396609-a7d2dadc-8162-11e8-9bf7-f50a7f7c57a0.png">

So also in TasmoAdmin it shows as off line. When going to it's IP address the sonoff can be toggled! But keeps telling the issue shown above.

arendst, 2018-07-06:
Status 0 output?

HFeenstra, 2018-07-06:
local_ip_address/cm?cmnd=status%200

File Not Found

URI: /cm
Method: GET
Arguments: 1
 cmnd: status 0

arendst, 2018-07-06:
Use console and execute command status 0

HFeenstra, 2018-07-06:
21:18:39 SRC: WebConsole from 192.168.1.240
21:18:39 RSL: Received Topic /command, Data Size 8, Data status 0
21:18:39 RSL: Group 0, Index 1, Command COMMAND, Data status 0
21:18:39 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}

arendst, 2018-07-06:
Only status 0

HFeenstra, 2018-07-06:
Sorry Arendst, my NOOB knownledge, but just updated above

jjlawren, 2018-07-06:
Looks like a problem with TasmoAdmin that I reported over here: https://github.com/reloxx13/TasmoAdmin/issues/165

HFeenstra, 2018-07-06:
I followed the executes you requested me to do:

otaurl "http://localip:port/data/firmwares/sonoff.bin"
upload 1
No success:
stat/sonoff/UPGRADE = Failed HTTP error: connection refused

Wait for tomorrow ;)

Thanks @jjlawren

jjlawren, 2018-07-06:
Check your OTA URL. TasmoAdmin seems to use `http://<your_ip>:<your_port>/data/firmwares/sonoff-full.bin`

arendst, 2018-07-06:
Just released the fix as v6.1.0 (again). wait a few minutes for the binaries to pop up.

HFeenstra, 2018-07-06:
Wow, that is fast...

So I tried the command:
otaurl "http://localip:port/data/firmwares/sonoff-full.bin"
upload 1

And then I see:
Failed HTTP error: connection refused

Well I will try to see what I need to learn how to execute it in the right way

arendst, 2018-07-06:
Where do you execute these commands? From the device webpage or from TasmoAdmin?

arendst, 2018-07-06:
In your case I guess it's ``otaurl http://192.168.1.111:9541/data/firmwares/sonoff.bin``

HFeenstra, 2018-07-06:
> In your case I guess it's otaurl http://192.168.1.111:9541/
> data/firmwares/sonoff.bin

From the device webpage. Ok let me try your command

HFeenstra, 2018-07-06:
Failed file not found gives it as a return.

So I tried ota http://..../sonoff-full.bin (in stead of sonoff.bin)

That did the trick! Thank you @Arendst ! It also came up in the TasmoAdmin docker again; as version 6.1.0

HFeenstra, 2018-07-07:
It works; to close this topic this is what I did to get it upgrading to full:

otaurl http://localip:port/data/firmwares/sonoff-full.bin
upload 1

It rebooted successfully and came back as with it's full functionality. Thanks.

closed 2018-07-07 by HFeenstra