arendst/Tasmota #3145: IR Send Command On Power On / Reboot Device

prasad7676, 2018-07-06:
I have HA Configuration 

- platform: mqtt
  name: "TV"
  state_topic: "tele/TV/RESULT" 
  command_topic: "cmnd/TV/irsend"
  payload_on: '{"Protocol": "NEC","Bits": 32,"Data": 551489775 }'
  payload_off: '{"Protocol": "NEC","Bits": 32,"Data": 551489775 }'
  optimistic: true
  retain: false
  qos: 0
 

every time device reboot it send ir command. is anyway to stop that in power on.
i tried poweronstate 3 but not working.

ascillato, 2018-07-06:
Please, share all the information of the troubleshooting template in orden to be able to help you:

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-07-10:
Hi,

As there is no feedback, I'm closing the issue. Please, feel free to reopen it and fill all the required information. We are willing to help you. Thanks.

closed 2018-07-10 by ascillato2