arendst/Tasmota #3147: Wemos D1 Mini and HC SR501 PIR Motion Sensor

hvddrift, 2018-07-06:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

When I boot my D1 Mini flashed with Tasmota is get this

00:33:00 MQT: tele/pirstairs/STATE = {"Time":"2018-07-07T00:33:00","Uptime":"0T00:00:15","Vcc":2.810,"Wifi":{"AP":1,"SSId":"vanderdriftfamily","RSSI":100,"APMac":"A2:3D:CF:FA:B6:AF"}}
00:33:00 MQT: tele/pirstairs/SENSOR = {"Time":"2018-07-07T00:33:00","Switch2":"OFF"}
00:33:05 MQT: cmnd/pirstairs/POWER = ON
00:33:05 MQT: stat/pirstairs/RESULT = {"Command":"Unknown"}

I have flashed it with Tasmota. And I used @Drzzs  method from this https://www.youtube.com/watch?v=sVml02kP3DU

When I motion the PIR the tasmota console outputs:

00:24:16 MQT: cmnd/pirstairs/POWER = ON
00:24:16 MQT: stat/pirstairs/RESULT = {"Command":"Unknown"}

Settings on D1 Mini
[image]

00:43:00 CMD: switchmode2
00:43:00 MQT: stat/pirstairs/RESULT = {"SwitchMode2":0}

------
Hassio config

binary_sensor:
  - platform: mqtt
    state_topic: "cmnd/pirstairs/POWER2"
    name: "D1 PIR"

Any suggestions would be appreciated.

```
00:36:38 MQT: stat/pirstairs/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"pirstairs","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:36:38 MQT: stat/pirstairs/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:53","StartupUTC":"2018-07-06T23:32:45","Sleep":0,"BootCount":65,"SaveCount":123,"SaveAddress":"F6000"}}
00:36:38 MQT: stat/pirstairs/STATUS2 = {"StatusFWR":{"Version":"6.1.0","BuildDateTime":"2018-07-06T21:01:47","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:36:38 MQT: stat/pirstairs/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vanderdriftfamily",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:36:38 MQT: stat/pirstairs/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","23B6179E","00000000"]}}
00:36:38 MQT: stat/pirstairs/STATUS5 = {"StatusNET":{"Hostname":"pirstairs-4829","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:1A:32:DD","Webserver":2,"WifiConfig":3}}
00:36:38 MQT: stat/pirstairs/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.38","MqttPort":1883,"MqttClientMask":"pir01","MqttClient":"pir01","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:36:38 MQT: stat/pirstairs/STATUS7 = {"StatusTIM":{"UTC":"Fri Jul 06 23:36:38 2018","Local":"Sat Jul 07 00:36:38 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:55","Sunset":"20:55"}}
00:36:38 MQT: stat/pirstairs/STATUS10 = {"StatusSNS":{"Time":"2018-07-07T00:36:38","Switch2":"ON"}}
00:36:38 MQT: stat/pirstairs/STATUS11 = {"StatusSTS":{"Time":"2018-07-07T00:36:38","Uptime":"0T00:03:53","Vcc":2.810,"Wifi":{"AP":1,"SSId":"apple","RSSI":100,"APMac":"A2:3D:CF:FA:B6:AF"}}}
```

00:38:05 MQT: tele/pirstairs/STATE = {"Time":"2018-07-07T00:38:05","Uptime":"0T00:05:20","Vcc":2.809,"Wifi":{"AP":1,"SSId":"vanderdriftfamily","RSSI":100,"APMac":"A2:3D:CF:FA:B6:AF"}}
00:38:05 MQT: tele/pirstairs/SENSOR = {"Time":"2018-07-07T00:38:05","Switch2":"ON"}

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-07-07:
Take a look at #1567  Do a search for PIR in issues you will find the answer you need.

ascillato2, 2018-07-10:
Hi,

Could you manage to solve your issue?

ascillato2, 2018-07-16:
Hi

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2