arendst/Tasmota #3149: SONOFF T1 3ch Odd Power Cycling After "Successful" Flash

BTLOPER, 2018-07-07:
Howdy all,

I am having an issue with a T1 3ch.  Obviously in trouble shooting, I had to work backwards eliminating variables, but the results are as follows.

BLUF:

MOD(c) = SONOFF T1 (represent the physical devices)
c =Number of channels (eg. 3 = SONOFF T1 3ch)

TASMOTA(build) = The framework we all know and love.
norm = My Normal Build as described below (Basically vanilla TASMOTA with just my wifi details)
min= My Minimal Build(with wifi details)
vmin= Vanilla Minimal Build from [https://github.com/arendst/Sonoff-Tasmota/releases]
vnorm= Vanilla Minimal Build from [https://github.com/arendst/Sonoff-Tasmota/releases]
up = Upgraded the firmware from min via WebUI (so not a firmware I built)

Config(m) = configured the module through the web UI to be a 
m = Module configuration (# correspond to the number of channels as above)

AOK = Module configures and work exactly as expected without any power cycling

PROBLEM(n, t) = Power for the relays cycles on and off the led touch pad turns on and off with the relay.
n = the number of relays cycling
s = seconds between power cycle events (either shown as [Array] of durations or ~s as an average across all channels)

Observations (in demonstrative order, not chronological)
MOD(1) + TASMOTA(norm) + Config(1Ch)  = AOK
MOD(2) + TASMOTA(norm) + Config(2 ch)  = AOK
MOD(3) + TASMOTA(min) = PROBLEM(1, ~55)
MOD(3) + TASMOTA(up) = PROBLEM(1, [300, 120, 24, 150, 40])
MOD(3) + TASMOTA(up) + Config(1) = PROBLEM(1, [60, 10, 26, 60, 23, 30])
MOD(3) + TASMOTA(up) + Config(1 + MQTT) = PROBLEM(1, [10, 26, 33, 40])
MOD(3) + TASMOTA(up) + Config(2 + MQTT) = PROBLEM(2, ~15)  (See notes on cycle order below) 
MOD(3) + TASMOTA(up) + Config(3 + MQTT) = PROBLEM(3, ~15)
MOD(3) + TASMOTA(up) + Config(Basic + MQTT) = PROBLEM(1, [975, 40, 25 +240])  (first cycle 9m35s after boot)
MOD(3) + TASMOTA(vnorm) + Config(3 + MQTT) = PROBLEM(3, ~15)
MOD(3) + TASMOTA(norm) =PROBLEM(1, [612, 93, 89, 82])
MOD(3) + TASMOTA(norm) + Config(MQTT) =PROBLEM(1,~30)
MOD(3) + TASMOTA(norm) + Config(3 + MQTT) =PROBLEM(1,~15) (this state is the goal and where I started)
MOD(3) + Example ESP8266 Sketches = OK (although blink or a simple web server are far from the goal)

Now for the long winded version:
I use VS Code and PlatformIO on Windows 10.  I have successfully flashed a 1ch and 2ch, but when it comes to the 3ch, I'm stuck, mind you I only have one some I can't rule out a hardware problem.  My standard build is basically out-of-the-box (or zip) TASMOTA v 5.14.  All I add for standard and minimal builds is the wifi details.  Almost 100% of the time I can successfully flash TASMOTA minimal firmware to the T1 3ch from PlatformIO.  Flashing the full build achieves only a 10% success rate.  Admittedly the the wiring was initially a little dicey and precariously holding a jumper on R19 to bring GPIO low to initiate flashing could be reducing success rate some (disproved later).  
The troubleshooting page suggests the the on/ off cycling is the result of a bad flash; so I've been working under that assumption trying various flashing methods as detailed above.  Here is a more verbose explaintion of my trials (and tribulations - most those).

1. Initially, and still not completely ruled out I was concerned about power.  I thought with all the LEDs and wifi cranking, maybe it was drawing two much current (at least during flashing) for the usb port, despite using a powered hub.  So I set up a dedicated power supply giving it a steady 3.3v up to 4.0A.  This had no impact on the results other than it increased my confidence in the connections as I re-soldered everything and set up an as simple-as-possible GPIO 0 low setup.
2. Then I turned to the firmware thinking the standard firmware might be too big.  Not necessarily too big to fit, but if there were random issues during flashing, reducing the time to flash might help.  So, I commented out the section about sensors, I have not considered what the T1 uses to detect touches, but for the time being, I'm just trying to stop the cycling of the pins (i.e. get a truly successful flash).  No dice here either.  The switches still cycle on and off; although, there seems to be an interesting trend in the cycle timing, more later.  
3. I have seen people say they have had success flashing from the Arduino IDE using the ESP8265 board setting, I saw a post from @arendst that the user should simply stick with the 8266 version; so for now, I haven't explored that.  
4. Following what seems to be a worst case guide in [troubleshooting,](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#relay-clicksled-flashes-at-1-sec-interval) I erased the flash via the ESPTool and re-flashed the MY standard firmware (over the next 12 hours I erased and flashed every firmware combination imaginable).  I didn't see any off the on/ off cycling (but I might not have waited long enough so after a bunch of other troubleshooting, I'm going to go back a verify that. See #18).  Thinking things were good, I proceeded with configuring in to be a T1 3CH module.  As soon as it came back from reboot, the cycling commenced.  
5-16.  Tried various iterations or the above in terms of reseting the CFG and not, and starting from minimal and not, and flashing via ESPTool and PlatformIO, all paths had the same results.  
17. Rule out plain ol' hardware problem.  I simply uploaded a few the ESP8266 example WebServer sketches from the Arduino IDE.  AOK!  Sketches all work, but most convincingly there is no power cycling which seems to suggest that its not a hardware problem... at least not a hardware problem that take TASMOTA to expose.
18. So finally, just to make sure all t's are dotted and i's are crossed (yeah, that sounds about right), I pretty much repeated all of that in reverse order to take the shorthand notes above.
19. I made a stiff drink! 

There is a key point regard which pad and relays are cycling.  the number of pads lighting and relays cycling is connected to the module configuration.  Basic and Ch1 one cycle just one pad Ch2 cycles two, and Ch3 cycles all three.  In my mind, that strongly points to software.  
There is also an interesting trend related to timing.  I provide the timing of the cycle events because they are clearly NOT completely arbitrary.  Part of it might be the inclusion of the 2nd and 3rd pad, as described just above (.i.e more pads, more opportunity for cycling events), but for sure, the simplest configs not only cycle slower they don't start cycling for a while after boot.  When upgrading from minimal firmware, the first cycle didn't happen for over 9 minutes.  Even changing the module configuration from T1 3ch down to Basic, the cycle showed a marked slowing and delay.
Finally, the cycling itself is completely irregular.  During on 2ch phase the cycles order was 1 2 1 1 2 1 1 1 1 2 2 2 2 2 2 2 2 2 2 1 2.  Unfortunately, that seems to point away from software.  
Second finally, I haven't really given and data because under test conditions, no results were repeatable, but I swear the move I was moving around, especially near the troubled board, the more frequent the cycle events.  Like I said, I couldn't prove it, but maybe EM interference is evolved.  Its just odd that no other boards have that problem.

Confusion (for lack of any Conclusion),
I haven't completely ruled out hardware as a contributing factor.  Since the 1 and 2 channel T1's are working perfectly with the same firmware, that does suggest something may be defective.  The odd timing of the cycle events also seems to lead away from software.  There maybe be some capacitors that are either bad or poorly soldered.  I'll look into that as much as I can, but there is the fact that any firmware other than TASMOTA doesn't suffer from cycling.  Therefore it is highly probable that something in the depths of TASMOTA is at least creating the condition for the cycling to occur.

If anybody has any suggestions, I'm all ears!  Thanks in advance.  Please forgive me, on the 4th proof reading pass, I just gave up!  

P.S. I should say that I have the PCB out of the case.  I don't think this would have much bearing, but its not actually triggering the relays.  I did confirm with some LEDs that the transistors are triggering the headers.  

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
[... cut ...]
20:17:46 RSL: stat/TASMOTA-T1-3-0001/RESULT = {"POWER":"OFF"}
20:17:46 RSL: stat/TASMOTA-T1-3-0001/POWER = OFF
20:17:49 MQT: Attempting connection...
20:17:49 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
20:18:00 MQT: Attempting connection...
20:18:00 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
20:18:11 RSL: stat/TASMOTA-T1-3-0001/RESULT = {"POWER":"ON"}
20:18:11 RSL: stat/TASMOTA-T1-3-0001/POWER = ON
20:18:11 MQT: Attempting connection...
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-07-07:
Try a build with arduino v.2.3.0. You use 2.4.1 -> known issues (not for all and always!)
See https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

BTLOPER, 2018-07-07:
Good suggestion.  Unfortunately results are the no different.

closed 2018-09-05 by stale[bot]

aitspk, 2018-12-24:
I flashed my T1 UK 3CH and unfortunately I ended up having the same power cycling issue tried various flashing method but it is not going any where

aitspk, 2018-12-24:
The power cycle issue solved once the relay board is connected back working normal on main

BTLOPER, 2018-12-25:
Confirmed.  I order and flashed a second 3-gang T1 and was sort of finicky on just usb power, but once I put it in the wall, it worked right as rain.  I'm guessing the first unit might work as well.   I don't really have a need for a second 3-gang though.  I guess I could just use it as an MQTT trigger linked via Home Assistant to my home's self-destruct mechanism?!?!???  Haha.   I'm glad you got yours working as well.

aitspk, 2018-12-25:
Yeah mine working now as well. :) thanks