arendst/Tasmota #3150: WiFi won't connect with 6.10

Stuart-Richards, 2018-07-07:
WiFi just does not connect with 6.10 firmware. If I connect console I see it saying it's trying to connect to 11N AP but then gives up. If I switch back to 5.14, all works just fine.

Sealight10, 2018-07-07:
I can't use my Wemos D1 Mini module with this tasmota version 6.1 because the connection turns very unstable. I don't have this trouble with the tasmota 5.14 version. 
I configured the option Wifi_manager in the wifi config tool (User_config.h), I'm wondering if there are any issue?
I checked my router and my module connects just for few seconds but afterward i can't get acces to the webserver. I used the tool ESPflashtool to clean the module before the flashing process.
I read about the posibility to use the libraries of the 5.14 version to compile the 6.1 version but after try it i got some errors. Could somebody else confirm if they have this troubles? Thanks in advance (And sorry for my english)

pascalsaul, 2018-07-07:
`stat/basic_woonkamer_bank/STATUS2 {"StatusFWR":{"Version":"6.1.0a","BuildDateTime":"2018-07-07T11:53:47","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}`

OTA upgrade invoked by the browser or MQTT doesn't work: stat/basic_woonkamer_bank/UPGRADE Failed Update error: ERROR[6]: Stream Read Timeout

Then I just upload the file manually and after the upgrade the WiFi connection looks fine but for me the webserver doesn't work anymore for me...

When I do a downgrade back to 5.14.0 throught MQTT it doesn't retain the config...

Jason2866, 2018-07-07:
Try the prebuilt image (release) from path Master not development.
Maybe your selfcompiled version is buggy...

bastelfix, 2018-07-07:
Hi, i have the same problem with iTead sonoff basic. 
PCB says "Sonoff Basic R2 V1.0 2017-10-11".
For me, the problem started with v5.14, I only use the precompiled DE-bin's. I have flashed also some of the "obisockets", a pow v2, some esp8266-01 and a node-mcu. All work fine with v5.14 and with the given wireless network (fritzbox, all dhcp leases cleared). 
But with the iTead basic, the last working version is v5.13
v5.11.1 and v5.13: working
v5.14 and v6.1: not working

I flash them using a ftdi-usb with esptool 2.4.1 with the following commands:
py esptool.py --port COM11 erase_region 0x0F4000 0x008000
py esptool.py --port COM11 erase_flash
py esptool.py --port COM11 write_flash -fs 1MB -fm dout 0x0 sonoff.bin

I can reproduce the behavior with at least six devices.

I also tried to flash the minimal-bin and then OTA, but the OTA fails.
 
Serial output (v6.1; seriallog 4)
`

00:13:16 APP: starte neu

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff38d8
~ld
ø
00:00:00 APP: Anzahl StartvorgÃ¤nge 6
00:00:00 SRC: Restart
00:00:00 Projekt sonoff Sonoff (Topic sonoff, Fallback DVES_A34C9C, GroupTopic sonoffs) Version 6.1.0-2_3_0
00:00:00 CFG: in Flash gespeichert am F6, zÃ¤hlen 46, Bytes 3584
00:00:00 WIF: PrÃ¼fe Verbindung...
00:00:00 WIF: Verbindungsversuch...
00:00:00 WIF: Repariere Problem #2186
00:00:00 WIF: verbinden mit AP1 **** in Modus 11N wie sonoff-3228...
00:00:01 WIF: PrÃ¼fe Verbindung...
00:00:01 WIF: Verbindungsversuch...
00:00:02 WIF: PrÃ¼fe Verbindung...
00:00:02 WIF: Verbindungsversuch...
00:00:03 WIF: PrÃ¼fe Verbindung...
00:00:03 WIF: Verbindungsversuch...
00:00:04 WIF: PrÃ¼fe Verbindung...
00:00:04 WIF: Verbindungsversuch...
00:00:05 WIF: PrÃ¼fe Verbindung...
00:00:05 WIF: Verbindungsversuch...
00:00:06 WIF: PrÃ¼fe Verbindung...
00:00:06 WIF: Verbindungsversuch...
00:00:07 WIF: PrÃ¼fe Verbindung...
00:00:07 WIF: Verbindungsversuch...
00:00:08 WIF: PrÃ¼fe Verbindung...
00:00:08 WIF: Verbindungsversuch...
00:00:09 WIF: PrÃ¼fe Verbindung...
00:00:09 WIF: Verbindungsversuch...
00:00:10 WIF: PrÃ¼fe Verbindung...
00:00:10 WIF: Verbindungsversuch...
00:00:11 WIF: PrÃ¼fe Verbindung...
00:00:11 WIF: Verbindungsversuch...
00:00:12 WIF: PrÃ¼fe Verbindung...
00:00:12 WIF: Verbindungsversuch...
00:00:13 WIF: PrÃ¼fe Verbindung...
00:00:13 WIF: Verbindung fehlgeschlagen, da AP nicht erreicht werden konnte
00:00:13 WIF: Repariere Problem #2186
00:00:14 WIF: verbinden mit AP1 **** in Modus 11N wie sonoff-3228...
00:00:14 WIF: PrÃ¼fe Verbindung...
00:00:14 WIF: Verbindungsversuch...
00:00:15 WIF: PrÃ¼fe Verbindung...
`
sometimes it drops

`
00:01:36 WIF: PrÃ¼fe Verbindung...
00:01:36 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
00:01:37 WIF: PrÃ¼fe Verbindung...
`

If i can provide any further information, please tell me 
If it is usefull, i can provide a device within germany.

Greets Bastelfix

PS: Yeah, first action at github :)

Jason2866, 2018-07-07:
@bastelfix 
Please post result of status 0. May you try not the german version for testing.
I have one NodeMcu which is working only until 5.13. But all my sonoffs do work with all versions including Sonoff Basic R2 V1.0 2017-10-11
Ich bin aus Deutschland...

Jason2866, 2018-07-07:
May you try with optioon for flashing speed
esptool.py --port COM11 erase_flash
esptool --port COM11 --baud 115200 write_flash -fs 1MB -fm dout 0x000000 firmware.bin

digiblur, 2018-07-07:
Flashed a S31, Dual R2, and Basic this morning with 6.1.0b built with core 2.3.0 in PlatformIO, no issues with WiFi connections so far.

bastelfix, 2018-07-07:
>May you try with optioon for flashing speed
esptool.py --port COM11 erase_flash
esptool --port COM11 --baud 115200 write_flash -fs 1MB -fm dout 0x000000 firmware.bin 

The --baud parameter didn't change anything
BUT
your advice to test with an english version was a key:
v5.13-DE: working

v5.14-DE: not working
v5.14: working

v6.1-DE: not working
v6.1: working

tested with two devices to avoid hardware-dependency

So maybe there is a bug from v5.13-DE to v5.14-DE that only shows up at the iTead sonoff basic?
Can you verify that with your device @Jason2866 ?
Did anyone else experienced this with the DE-bin?

Thanks so far,
Bastelfix
Berlin

Jason2866, 2018-07-07:
@bastelfix 
Do you have changed wifi settings in user_config.h to your setup?

bastelfix, 2018-07-07:
no, i used precompiled release-bins

Jason2866, 2018-07-07:
Do you have read the wiki how establishing to wifi works?
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

bastelfix, 2018-07-07:
of course, but i configure the wifi via serial (ftdi-usb) with the backlog command:

Backlog SSID1 netssid; Password1 pw

00:00:12 CMD: Backlog SSID1 netssid; Password1 pw
00:00:12 RSL: stat/sonoff/RESULT = {"SSId1":"****"}
00:00:12 WIF: Connect failed with AP timeout
00:00:12 WIF: Connecting to AP1 StonedNet_n in mode 11N as sonoff-4402...
00:00:12 RSL: stat/sonoff/RESULT = {"Password1":"pw"}
00:00:14 APP: Restarting

then it tries to connect 

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_86B132, GroupTopic sonoffs) Version 6.1.0-2_3_0
00:00:00 WIF: Connecting to AP1 *** in mode 11N as sonoff-4402...
00:00:10 WIF: Connected
00:00:10 DNS: Initialized
00:00:10 HTP: Web server active on sonoff-4402.local with IP address 192.168.****

jjlawren, 2018-07-07:
I was having a similar issue with two Basics with the precompiled 6.1.0 release binary. Both had been on a locally built 5.13.1 with a custom `user_config_override.h` before the upgrade. After the upgrade to 6.1.0, one simply wouldn't connect to wifi and the other's wifi became unstable, losing connection frequently. The uptime kept increasing when it published status updates, so it was not rebooting.

After rolling both back to a locally built 5.13.1 w/2.4.1 things seem stable again.

Jason2866, 2018-07-07:
@bastelfix 
Compiled (in der Halbzeitpause) v.6.1.0a development  (latest) and flashed it to my easy reachable sonoff... No problem. Will build de version after soccer Rus - Cro and flash it afterwards.

00:00:00 Project sonoff Fluter (Topic sonoff-13D92F, Fallback sonoff-13D92F, GroupTopic sonoffs) Version 6.1.0a-2_3_0
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-13D92F-6447...
20:55:48 WIF: Connected
20:55:48 HTP: Web server active on sonoff-13D92F-6447 with IP address 192.168.2.119
20:55:48 UPP: Multicast (re)joined
20:55:48 MQT: Attempting connection...
20:55:48 MQT: Connected
20:55:49 MQT: tele/sonoff-13D92F/LWT = Online (retained)
20:55:49 MQT: cmnd/sonoff-13D92F/POWER = 
20:55:49 MQT: tele/sonoff-13D92F/INFO1 = {"Module":"Sonoff S2X","Version":"6.1.0a","FallbackTopic":"sonoff-13D92F","GroupTopic":"sonoffs"}
20:55:49 MQT: tele/sonoff-13D92F/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-13D92F-6447","IPAddress":"192.168.2.119"}
20:55:49 MQT: tele/sonoff-13D92F/INFO3 = {"RestartReason":"Software/System restart"}
20:55:49 UPP: Multicast (re)joined
20:55:50 MQT: stat/sonoff-13D92F/RESULT = {"POWER":"OFF"}
20:55:50 MQT: stat/sonoff-13D92F/POWER = OFF
20:55:53 UPL: File Tasmota_dev_6.1.0a.bin ...
20:56:00 UPP: Multicast (re)joined
20:56:00 MQT: Attempting connection...
20:56:00 MQT: Connected
20:56:01 MQT: tele/sonoff-13D92F/LWT = Online (retained)
20:56:01 MQT: cmnd/sonoff-13D92F/POWER = 
20:56:01 UPP: Multicast (re)joined
20:56:03 MQT: stat/sonoff-13D92F/RESULT = {"POWER":"ON"}
20:56:03 MQT: stat/sonoff-13D92F/POWER = ON
20:56:03 MQT: stat/sonoff-13D92F/RESULT = {"POWER":"OFF"}
20:56:03 MQT: stat/sonoff-13D92F/POWER = OFF
20:56:04 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-07-07T20:56:04","Uptime":"0T00:00:26","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":96,"APMac":"00:A0:57:2A:BD:19"}}
20:57:06 MQT: stat/sonoff-13D92F/RESULT = {"POWER":"ON"}
20:57:06 MQT: stat/sonoff-13D92F/POWER = ON
20:57:07 MQT: stat/sonoff-13D92F/RESULT = {"POWER":"OFF"}
20:57:07 MQT: stat/sonoff-13D92F/POWER = OFF
21:01:05 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-07-07T21:01:05","Uptime":"0T00:05:27","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":90,"APMac":"00:A0:57:2A:BD:19"}}
21:02:00 MQT: tele/sonoff-13D92F/UPTIME = {"Time":"2018-07-07T21:02:00","Uptime":"0T00:06:22"}
21:06:05 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-07-07T21:06:05","Uptime":"0T00:10:27","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":92,"APMac":"00:A0:57:2A:BD:19"}}
21:11:06 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-07-07T21:11:06","Uptime":"0T00:15:28","Vcc":3.127,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
21:11:22 CMD: Status 0
21:11:22 MQT: stat/sonoff-13D92F/STATUS = {"Status":{"Module":8,"FriendlyName":["Fluter"],"Topic":"sonoff-13D92F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/tasmota/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:15:44","StartupUTC":"2018-07-07T18:55:38","Sleep":1,"BootCount":18,"SaveCount":209,"SaveAddress":"FB000"}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS2 = {"StatusFWR":{"Version":"6.1.0a","BuildDateTime":"2018-07-07T20:01:00","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"SetOption":["00008009","55C18000"]}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS4 = {"StatusMEM":{"ProgramSize":512,"Free":488,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FAAA78C","0C000001","00080696","00000000"]}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS5 = {"StatusNET":{"Hostname":"sonoff-13D92F-6447","IPAddress":"192.168.2.119","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"EC:FA:BC:13:D9:2F","Webserver":2,"WifiConfig":3}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-13D92F","MqttClient":"sonoff-13D92F","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 07 19:11:22 2018","Local":"Sat Jul 07 21:11:22 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"05:18","Sunset":"21:17"}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS10 = {"StatusSNS":{"Time":"2018-07-07T21:11:22"}}
21:11:22 MQT: stat/sonoff-13D92F/STATUS11 = {"StatusSTS":{"Time":"2018-07-07T21:11:22","Uptime":"0T00:15:44","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}}

Jason2866, 2018-07-07:
Info. De version 6.1.0a works too. Self compiled and @reloxx13 clean de build

reloxx13, 2018-07-07:
i remeber we had problems with a german release in the past, maybe theos IDE or system does have a charset problem.

`00:01:10 RSL: empfangenes topic /status, DatengrÃ¶ÃŸe 1, Daten 0`

i dont have those charset errors with my builds.

rkojedzinszky, 2018-07-07:
I also had that issue, even with fresh flashing. It turned out for me that I had to undef USE_ENERGY_SENSOR for my Sonoff Basics to work with master branch. My other Sonoff devices are not affected, only the Basics.

reloxx13, 2018-07-08:
also dont forget that wifi relevant change short  mentioned here: https://github.com/arendst/Sonoff-Tasmota/issues/3101

if you build yourself and want to debug, revert the line in your local code, rebuild and check if your connection gets stable again.

bastelfix, 2018-07-08:
> Jason2866 commented 15 hours ago
Info. De version 6.1.0a works too.

The One from releases or a self-compiled one? Can you please provide a link? Then i can test with my devices. 

What about the Migration Path? I thought it is only relevant if i'm upgrading from version <5 via OTA. But I flash them via serial, so this shouldnt be an issue?

At the moment i'm happy with the the english build, but it would be nice if anybody can reproduce this behaviour with DE-v5.14 or DE-v6.1 at the "Sonoff Basic R2 V1.0 2017-10-11"-Hardware so i can exclude an error made by me. 
(Again, ) DE-5.14 is working fine on "obisocket", esp8266-01, pow v2 and nodeMCU for me, only the basics fail to connect to wifi.  

Thanks for all the support to all of you and please execuse my dirty english :)

Jason2866, 2018-07-08:
I tried @reloxx13 and my de version, which works. reloxx13 has uploaded this files 6 or 7 posts before

bastelfix, 2018-07-08:
nice, working!
v5.13-DE: working

v5.14-DE: not working
v5.14: working

v6.1-DE: not working
v6.1-DE-reloxx13: working, connects faster (4s to 7s) than v6.1
v6.1: working, connection time 4s to 34s

> If you upgrade take a look how to migrate from v.5.14. You have to care!
That should be sticky in a way as reloxx13 mentioned

ChipStewart1, 2018-07-09:
I'm having similar problems.  I can flash 5.14.0b and everything works fine.  When I upgrade to 6.1, however, it won't connect.  On the serial monitor, I get:

00:00:00 Project sonoff Lamp (Topic Lamp, Fallback Lamp, GroupTopic Home) Version 6.1.0a-2_4_1
00:00:00 WIF: Connecting to AP1 The Dog House in mode 11N as Lamp-3447...
00:00:09 WIF: Connect failed as AP cannot be reached
00:00:09 WIF: Connecting to AP1 The Dog House in mode 11N as Lamp-3447...
00:00:18 WIF: Connect failed as AP cannot be reached
00:00:18 WIF: WPSConfig active for 3 minutes
00:00:41 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
[which repeats indefinitely ]

I tried via updating via the web interface, both via .bin file and the OTA link.  I tried a clean install.  I tried the regular and minimal builds.  Still, nothing seems to work.

Jason2866, 2018-07-09:
@ChipStewart1 Try to use arduino 2.3.0! For me 2.4.x has never worked...

ChipStewart1, 2018-07-09:
2.3?  I’m on 1.85, and 1.9 is a beta.

 

And it seems to be uploading okay – just not connecting to WiFi afterwards.  And even the OTA isn’t working for me.

 

Thanks,

 

Chip Stewart

reloxx13, 2018-07-09:
you are on 2.4.1, i dunno wher eyou get the numbers 1.85...

00:00:00 Project sonoff Lamp (Topic Lamp, Fallback Lamp, GroupTopic Home) Version 6.1.0a**_-2_4_1_**

digiblur, 2018-07-09:
Confusing the compiler and library versions folks. In ArduinoIDE it's the board manager section where 2.30 or 2.41 is installed.

Jason2866, 2018-07-09:
@digiblur I dont know the actual Arduino Ide Version
I dont use it anymore since i "found" Platformio (Atom) and VSC.
It is so much easier to use even for me as non programmer.
Just follow the perfect tutorials in wiki from tasmota how to setup this IDE.
Compiling is done so easy. No hassle with files to copy somewhere and so on.
Just download project from github extract to folder, "open folder" in VSC and hit button compile.
Platform changes are all in platformio.ini... (This file is preconfigured from @arendst )

Jason2866, 2018-07-09:
@digiblur sorry my fault. Infos is for @ChipStewart1

helmar74, 2018-07-10:
Hi, 
I had the same issue. All sonoffs (S20, 433 MHz, 4CH Pro, SC, B1...) are working well with Stable Varsion 6.1.0 and german translation

...Except the sonoff Basic. It is possible to update to minimal version 6.1.0 via OTA, but Update to "normal" version isn't possible. Device was stuck.
That's how i was able to fix:
1. Used Development Version 6.1.0a (Stable version 6.1.0 isn't working for me)
2. Don't use german translation (if used, flashing via serial connections works, but no wifi connection is possible):
     Commented out the following in user_config.h:
     //#define MY_LANGUAGE            de-DE           // German in Germany
3. with this change I was able to update via serial connection
4. OTA Update to minimal version and then back to "normal" version isn't possible at all

Regards

Helmar

apastuszak, 2018-07-10:
So, am I reading this right?  Right now, there is no way to OTA a Sonoff Basic?

arendst, 2018-07-10:
@apastuszak No you're not reading it right. People seem to have problems with OTA. I don't, many others do not either. So.....

apastuszak, 2018-07-10:
It just failed on me on a Sonoff Basic.  Need to disassemble and flash via serial.

arendst, 2018-07-10:
@apastuszak if possible pls provide output of ``status 0``

apastuszak, 2018-07-10:
I can't connect to the device.  Comes up in Wifi Manager, but attempts to connect to the AP fail.  I held the button down for 40 seconds to reset the device to defaults, and same issue.  4 quick clicks to get it into wifi_manager and I an AP, but can't connect to it.

The last version I was successfully able to upgrade via wifi was 5.13.0

arendst, 2018-07-10:
OK. onto to the serial upload then.

apastuszak, 2018-07-10:
Just annoying since the basics are hot glued down into 3d printed boxes a friend made for me.

jjlawren, 2018-07-10:
> Comes up in Wifi Manager, but attempts to connect to the AP fail. I held the button down for 40 seconds to reset the device to defaults, and same issue. 4 quick clicks to get it into wifi_manager and I an AP, but can't connect to it.

I had the same behavior with one of my two Basics. Console showed repeated wifi failed connection attempts (after I hooked to serial cable), and attempting to join to the Basic in AP mode would time out.

I had upgraded from a custom 5.13.1 build to the 6.1.0 release build binary. My build was prettty minimal but did have a custom timezone override. Could a sudden timezone change + something stored in RTC memory cause wifi connection issues because of a time-based dependency?

helmar74, 2018-07-10:
Fyi:
Had the same problem, using timezone99...

apastuszak, 2018-07-10:
I have a custom timezone.  But when I went from 5.13 to 5.14 I did not have a custom timezone and still have to go serial.

That does seem to be something we have in common.  May be worth investigating.

arendst, 2018-07-10:
Starting with 5.14 the timezone settings in user_config.h have changed. See releasenotes and compare user_config.h between both versions.

apastuszak, 2018-07-10:
Would that cause wifi upgrade issues?

arendst, 2018-07-10:
I don't think so but it explains peoples timezone issues.

ChipStewart1, 2018-07-12:
Strange.  I’m working with two Sonoff Basics that I bought together.  One, I was able to flash in 5.14, then reflash in 6.1.0a.  I used the 40 second hold followed by the four quick taps as mentioned by apastuszak, and was able to access the network it set up and get to the web server interface, which I then used to set everything up.  It’s working fine now.

 

The other one – not so much.  I flash in 5.14 and everything works fine.  I flash in 6.1 (I tried normal and minimal) and it can’t access the APs.  I use the 40/4 method, and it says it set up the web server, but my phone isn’t showing the network.  I tried WifiConfig 1, 2, and 3, but no luck.

 

Chip

arendst, 2018-07-12:
My latest basic has writings: Sonoff TH_V1.1 2017-5-5

apastuszak, 2018-07-12:
Mine is a Sonoff Basic R2
v 1.0
10-11-2017

digiblur, 2018-07-12:
Hmm...so food for thought, I have to do some additional testing here though.  I got a few NodeMCUs in the other day with 4 meg flash chips.  I tested them like I normally do and one wouldn't connect to the WiFi at all.  I use a little test sketch the 2.3.0 core that just connects to WiFi so I can ping it and then I test a few GPIO in/out pins.  I chunked it in the drawer as dud and maybe for a project that doesn't need WiFi one day.  I even flashed Tasmota 6.x (don't remember which one at the time) that is on my other various Sonoff switches.  No dice.

With this thread I decided to try something, I flashed ESPEasy on it and boom, she was alive with WiFi!  So about 12 hours ago I fired up the latest development repo and compiled and flashed it.   WiFi still alive!  Nice!  So I popped it in my breadboard and fired it up with a couple i2c sensors, the wifi was dead.  I tried all sorts of things, popped it out of the breadboard, etc, reflashed, nothing.   Still dead.  I then flashed ESPEasy on it again while it was still sitting there on the table and boom, WiFi is alive again.  Very strange!  

I'm thinking something is faulty with this chip though as it did work a couple times with Tasmota, then just died when I put it int the breadboard.  It's odd that ESPEasy has worked every time though.  I think I'm going to try my little test sketch again and see if it is a core 2.30/2.4.1 thing maybe, try to narrow this down type deal.  Hoping I can get it to break with ESPEasy so I can ultimately rule that this is just a faulty chip.  

For the record though, I don't have WiFi issues with any of my S31s, Duals, Basics, Wemos D1 minis, etc running 6.x of Tasmota built with platformIo 2.3.0 core.

arendst, 2018-07-12:
Comparing the two reveals major change in power supply design but that seems to be the only change. They still use the ESP8266

Jason2866, 2018-07-12:
@digiblur  Sensors on a breadboard can cause different undefined troubles! 
I had bad experiences with that. Device does not react (seems dead) laggy, Wifi problems!!
Now i only connect everything direct with short lines -> no problems

apastuszak, 2018-07-12:
A small update.  I tried to upgrade an older Sonoff from 5.1.2 to 5.1.4 and
had all the exact same issues, except I could connect via WIFI_MANAGER.  I
went to my router and created a new 2.4 Ghz only network and gave it a 16
character password that only container letter and numbers, and was able to
get a good wifi signal.  I don't have any more time to tinker today.
Tonight I will try and upgrade a Sonoff Basic from 5.1.2 to 6.1.0 using my
new wifi network and see how it goes.

digiblur, 2018-07-12:
@Jason2866 right, sensors can do odd things, but when they've been running fine on the same NodeMCU type and even some Wemos D1s for a couple weeks now I don't have any reason to question them.

apastuszak, 2018-07-12:
The switch that gave me issues going from 5.12 to 5.14 that was corrected by a new 2.4 Ghz SSID with a password that was just letters and numbers was my guinea pig.

I flashed that with sonoff-minimal.bin from Github.  It rebooted and now has a blinking green light.  If I put it in WIFI_MANAGER, then I see the SSID, but my iPhone gives me an error and says it can't connect.

apastuszak, 2018-07-12:
Final update.  Used ESPtool to erase the SOnoff Basic.  Then I flashed 6.1 from Github and I can't get it to come up now.  Fires up.  I can put it in WIFI_MANAGER, but I can't connect to the AP it creates.

Is there anything useful I can do with it to help troubleshoot the issue?

apastuszak, 2018-07-12:
@arendst I found a Sonoff Basic TH_1.1.  I'll try and take that to 6.1 via OTA and see what happens.

apastuszak, 2018-07-12:
@arendst The TH_1.1 Sonoff Basic upgraded to 6.1.0 OTA without any issues using the minimal+full method.  I'm on my old wifi with the special character in it and it seems to be working fine.

digiblur, 2018-07-12:
Well count my weird NodeMCU chip out, I can't seem to narrow it down.  Definitely not a 2.30/2.41 issue or 5.x vs 6.x issue, so exclude my testing out the mix.  I believe I've found the issue with my NodeMCU, updated from 5.14 to 6.01a via OTA without issue and then I saw it...  bet this has an big deal with my flaky WiFi:  "Vcc":2.836

mike2nl, 2018-07-14:
Hi all,
i don't know if this is any solution but i had a lot of WiFi issues with the BASIC (R2) models.
Never with the S20 or any other iTead type. 
After a short time or from up the bginning i could'nt reach the config page or the main page.
On my router (Linksys WRT1900AC) on the browse page the BASIC was visible, strange thing.

Then i rembered that i had read a lot of sOnOff and Tasmota and i had something about
WIFI_SLEEP in my head.

I use:
- Arduion: 1.8.5
- Firmware: Sonoff-Tasmota-6.1.0 latest

What i have found in support.ino
#ifndef ARDUINO_ESP8266_RELEASE_2_4_1     
// See https://github.com/arendst/Sonoff-Tasmota/issues/2559
    WiFi.setSleepMode(WIFI_LIGHT_SLEEP);  // Allow light sleep during idle times
#endif

That i have changed to (added the two slashes before WiFi.setSleepMode):
#ifndef ARDUINO_ESP8266_RELEASE_2_4_1     
// See https://github.com/arendst/Sonoff-Tasmota/issues/2559
    **//**WiFi.setSleepMode(WIFI_LIGHT_SLEEP);  // Allow light sleep during idle times
#endif

Now it works like charm. Had flashed right now 16 BASIC types with issues
and thay are all gone right now. All 16 working like never before.

It looks that the WiFi don't come back from the loop (sometimes or all the times).

Jason2866, 2018-07-14:
@meingraham 
What arduino core version do you use? 2.3.0 or a 2.4.x ?
If 2.4.x try 2.3.0.

meingraham, 2018-07-14:
@Jason2866 

Yes, I'm using 2.4.1.

I edited my platformio.ini to point to 2.3.0 and tried to compile 6.1.0b and my old "esptool" not found nemesis came back. I never was able to resolve that problem without a full reinstall of my development environment. I'm beginning to understand perhaps where that "esptool" error was coming from in that the linker can't find it in the 2.3.0 package path... but I don't know how to resolve that (either by updating the path or installing esptool in the proper 2.3.0 folder path). I thought esptool was used for uploading but apparently it is used to link the image as well.

Net-net... I'd like to test 6.1.0b with 2.3.0 just as a matter of confirming the Wi-Fi issue... and also so I can use the "no WPS" option. But, alas, until 6.1.0b is "stable" with 2.4.1 or I can get someone to give me guidance on resolving "esptool", I'm sticking with 6.1.0a.

Regards.

Mike

apastuszak, 2018-07-15:
Ok, the Sonoff Basic R2 that refused to work, I got working!  I recompiled using Arduino Core 2.4.1 and then used esptool to erase_flash and then flash my compiled image that was compiled with 2.4.1.  WIFI_MANAGER came up.  I connected and put in my wifi info.  The Sonoff rebooted and immediately connected to my wifi without issues.

Jason2866, 2018-07-15:
@meingraham  @apastuszak 
Long time ago i had problems with mDNS -> laggy devices and slow wifi.
I dont need it, so i disabled it by default

[24 more replies]

closed 2018-08-25 by ascillato2