arendst/Tasmota #3151: 6.1.0 GPIO Module Parameters for Switchs and Relays number changed -> caused incompatible issue.

wongnam, 2018-07-07:
Hi Arendst,

i want to suggested that in the future if you want to add more Switchs/Relays or any kind of module please don't change the number of them in the new FW version due to all our config for the sonoff is using backlog command with switch/ relay number to assign GPIO.
Now you already change them in new FW 6.1.0 will caused my Sonoff with different FW version incompatible to my system configuration. Thanks.

[image]

[image]

arendst, 2018-07-07:
You're right. I had a request for more switches a long time ago and knew this to be one of the consequences.

It was either no additional switches, messy additional switches or the solution I opted for.

Thx for the documentation and sorry for the inconvenience I may have caused.

wongnam, 2018-07-07:
I understand! i've just input my suggestion only.

In the case of no best option, it is important to decide with a method that we knows that it can cause some inconvenience, however we can accept it and alert the user in advance to avoid the potential issue before use.

Thank you.

closed 2018-07-07 by arendst