arendst/Tasmota #3152: Time triggered rule does not work?

kajmaj, 2018-07-07:
On sonoff basic, fw 6.0.0, rules activated, I created rule to publish value and delete counter every midnight
`on Time#Minute=001 do publish cmnd/rain/counter2 %value% endon on Time#Minute=001 do counter2 0 endon`
It does not work.
I tried to publish every minute
`on Time#Minute do publish cmnd/rain/counter2 %value% endon`
It does not work either.

arendst, 2018-07-07:
Just tested and works as expected:
```
rule2 on Time#Minute do publish cmnd/rain/counter2 %value% endon
10:12:59 CMD: rule2 on Time#Minute do publish cmnd/rain/counter2 %value% endon
10:12:59 SRC: Serial
10:12:59 RSL: Group 0, Index 2, Command RULE, Data on Time#Minute do publish cmnd/rain/counter2 %value% endon
10:12:59 MQT: stat/ring2/RESULT = {"Rule2":"OFF","Once":"OFF","Free":453,"Rules":"on Time#Minute do publish cmnd/rain/counter2 %value% endon"}
10:12:59 CFG: Saved to flash at F8, Count 119, Bytes 3584
rule2 1
10:13:06 CMD: rule2 1
10:13:06 SRC: Serial
10:13:06 RSL: Group 0, Index 2, Command RULE, Data 1
10:13:06 MQT: stat/ring2/RESULT = {"Rule2":"ON","Once":"OFF","Free":453,"Rules":"on Time#Minute do publish cmnd/rain/counter2 %value% endon"}
10:13:06 RUL: TIME#MINUTE performs "publish cmnd/rain/counter2 613"
10:13:06 SRC: Rule
10:13:06 RSL: Group 0, Index 1, Command PUBLISH, Data cmnd/rain/counter2 613
10:13:06 MQT: cmnd/rain/counter2 = 613
10:13:07 CFG: Saved to flash at F7, Count 120, Bytes 3584
10:14:00 RUL: TIME#MINUTE performs "publish cmnd/rain/counter2 614"
10:14:00 SRC: Rule
10:14:00 RSL: Group 0, Index 1, Command PUBLISH, Data cmnd/rain/counter2 614
10:14:00 MQT: cmnd/rain/counter2 = 614
```
Make sure to enable your rule with command ``rule 1``

kajmaj, 2018-07-07:
I have Rule2 enabled, but it still not MQT. 

C1 (Rule1) is connected to GPIO 14, Rule1 works as expected (Based on Tele-)
C2 (Rule2) is connected to GPIO , Rule2 does not work as it should. (may be this the reason?)
`
11:00:59 CMD: Rule2
11:00:59 SRC: WebConsole from 192.168.1.109
11:00:59 RSL: Received Topic /Rule2, Data Size 0, Data 
11:00:59 RSL: Group 0, Index 2, Command RULE, Data 
11:00:59 MQT: stat/sonoff_4/RESULT = {"Rule2":"ON","Once":"OFF","Free":453,"Rules":"on Time#Minute do publish cmnd/rain/counter2 %value% endon"}
11:01:03 MQT: tele/sonoff_4/STATE = {"Time":"2018-07-07T11:01:03","Uptime":"0T00:00:53","Vcc":3.231,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXX","RSSI":50,"APMac":"08:96:D7:2A:01:D7"}}
11:01:03 MQT: tele/sonoff_4/SENSOR = {"Time":"2018-07-07T11:01:03","COUNTER":{"C1":64,"C2":34}}
11:01:03 RUL: TELE-COUNTER#C1>0 performs "publish cmnd/wind/counter1 64"
11:01:03 SRC: Rule
11:01:03 RSL: Received Topic /publish, Data Size 21, Data cmnd/wind/counter1 64
11:01:03 RSL: Group 0, Index 1, Command PUBLISH, Data cmnd/wind/counter1 64
11:01:03 MQT: cmnd/wind/counter1 = 64
11:01:03 RUL: TELE-COUNTER#C1>0 performs "counter1 0"
11:01:03 SRC: Rule
11:01:03 RSL: Received Topic /counter1, Data Size 1, Data 0
11:01:03 RSL: Group 0, Index 1, Command COUNTER, Data 0
11:01:03 MQT: stat/sonoff_4/RESULT = {"Counter1":0}
'

kajmaj, 2018-07-07:
This rule : `Rule2 on Time#Minute do publish cmnd/rain/counter2 %value% endon` 
should publish STAT each minute I believe
This rule: `Rule2 on Time#Minute=001 do publish cmnd/rain/counter2 %value% endon on Time#Minute=001 do counter2 0 endon`
Should Publish STAT daily, after midnight and reset counter afterwards.
Or it is my misunderstood?
First rule2 I set up for testing only because I did not want to wait so long.

Status 0:
```
12:07:39 CMD: Status 0
12:07:39 MQT: stat/sonoff_4/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff_4"],"Topic":"sonoff_4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:07:39 MQT: stat/sonoff_4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Exception","Uptime":"0T00:03:37","StartupUTC":"2018-07-07T11:04:02","Sleep":50,"BootCount":3535,"SaveCount":3861,"SaveAddress":"FA000"}}
12:07:39 MQT: stat/sonoff_4/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-24T18:21:01","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:07:39 MQT: stat/sonoff_4/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HugoNET",""],"TelePeriod":10,"SetOption":["00008009","55818000"]}}
12:07:39 MQT: stat/sonoff_4/STATUS4 = {"StatusMEM":{"ProgramSize":491,"Free":512,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0DC02B94","0C000000","1402079A","00000000"]}}
12:07:39 MQT: stat/sonoff_4/STATUS5 = {"StatusNET":{"Hostname":"Sonoff_4","IPAddress":"192.168.1.64","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:7A:4B:98","Webserver":2,"WifiConfig":3}}
12:07:39 MQT: stat/sonoff_4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A4B98","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:07:39 MQT: stat/sonoff_4/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 07 11:07:39 2018","Local":"Sat Jul 07 12:07:39 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 02:00:00 2018","Timezone":1}}
12:07:39 MQT: stat/sonoff_4/STATUS10 = {"StatusSNS":{"Time":"2018-07-07T12:07:39","COUNTER":{"C1":8,"C2":34}}}
12:07:39 MQT: stat/sonoff_4/STATUS11 = {"StatusSTS":{"Time":"2018-07-07T12:07:39","Uptime":"0T00:03:37","Vcc":3.231,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HugoNET","RSSI":48,"APMac":"08:96:D7:2A:01:D8"}}}
```

kajmaj, 2018-07-07:
It is solved I guess. 
I was on 6.0 and Time# is on 6.1.
Thank you for guidance.

Closing the issue.

closed 2018-07-07 by kajmaj