arendst/Tasmota #3156: Theo's Tasmota Tips and new release v6.1.0

GasTurbineMan, 2018-07-07:
Again, and first of all - Tasmota is a great product for the Sonoff devices.  Great work by Theo.

I see where v6.1.0 as an official release and no longer in development mode.  Glad to see continued updates to the software.

Will "Theo's Tips" be updated as well?  I have read  "20180523 - Relation Tasmota and ESP8266/Arduino core version" a few weeks ago, but quite frankly, I could not understand how to implement the workarounds needed for core version 2.4.1, so I am still using core 2.3.0

If "workarounds" are still needed to run v6.1.0 with core 2.4.1; would it be possible to provide some clarification on how to install the needed code mods?
 
Thank you.

arendst, 2018-07-07:
I guess you refer to the Hue workaround as the other 2.4.1 workarounds are served by Tasmota. They are independent from Tasmota versions as they are introduced by the underlying core versions. I keep these workarounds in Tasmota as long as the current core versions are present.

Follow the link in the Hue item and read the issue log regarding where to fix the webserver problem in 2.4.1 core. You'll need some background knowledge to handle this or just refrain from 2.4.1 and keep using the released 2.3.0

GasTurbineMan, 2018-07-07:
I followed link and understood the workaround for the Hue issue – and was able to make the needed change.

 

The one I could not understand was the memory leak issue.  I followed the provided link – but it appeared to me there were 2 or maybe 3 different opinions on how to resolve the issue.  And none of them provided any specific instructions on what to change.  Was the memory leak issue on 2.4.1 resolved with v6.1.0?

arendst, 2018-07-07:
As I said before. The workarounds are needed BECAUSE THE UNDERLYING CORE LIBRARY contains errors like indeed memory leaks. 

As no new core library has been released the errors are still there and the workarounds are ALSO STILL PRESENT in Tasmota. No need for a user to change anything as the workarounds are in Tasmota.

reloxx13, 2018-07-07:
Note: dont forget that 2.4.1 still cannot sleep the wifi connection, which results in a higher power consumption.
If you care about your electricity bill, stay on 2.3.0.

Sleep is bugged in the Core, which lays even UNDER the UNDERLYING CORE LIBRARY of tasmota.

The core is managed by the chinese as much as i know. I expect a fix for this in a few months, since they changed alot api calls including the sleep api in the past months. Thats why i hope the sleep bug will be the next step. The latest released core with the api changes got already included in the latest arduino lib.

Tasmota -> Arduino ESP library -> ESP Core

GasTurbineMan, 2018-07-07:
OK, so you say "no need for user to change anything . . .".  I downloaded v6.1.0a and set PlatformIO.ini to use core 2.4.1 and set user_config.h for hue emulation.  Flashed Sonoff Basic and while Alexa discovers the new device, when I try and get Alexa to turn device on/off, I get the "device is unresponsive".

I went back with 2.3.0 and Alexa / Hue are working again.  I will stick with 2.3.0 for now until the core library errors are corrected.

Just now starting to work with temperature sensors.  Tasmota rocks!!

reloxx13, 2018-07-07:
If you build it yourself, you HAVE TO insert the HUE Workaround local on your system. As said, it cannot be fixed in the tasmota sourcecode.

The bugged code is automatic downloaded by your platformIO software as underlaying library and does not lie inside your tasmota directory.

Note: i dont know if theo included the hue fix in the ready to use bins.

arendst, 2018-07-07:
@reloxx13 thx. will add it to my release task list which already takes me half a day ...

reloxx13, 2018-07-07:
thats why i did not mention it before, took to much time for bugs which are not your fault.
and i dont know how it works in arduino ide, i just hate this ide so much 👎 

i dont know what takes it so long for you.

i dont get it why you dont use the merge from dev to master, i always see the master line get never a merge, all files in the master looking like they are getting updated by your hand.

arendst, 2018-07-07:
That's right. In my understanding I cannot merge dev to master because:

- readme is different
- platformio.ini is different
- some features in dev are not ready for master like display and/or sensor drivers
- I do not want debug file in master

If there is a way to merge while keeping the above in mind let me know...

reloxx13, 2018-07-07:
yup, the keyword is gitattributes
```
git config --global merge.ours.driver true

echo README.md merge=ours' >> .gitattributes
echo platformio.ini merge=ours' >> .gitattributes
```
this ways, on a merge always the files of the merged to branch gets used

but if you made changes to those files in dev, you need to make them manual in master

the features needs to be as a feature branch (git flow), their will no other way around i think
steps than woud be:

- normal dev in dev branch,
- now work on the new sensor:
  - commit everything in the dev branch or create a patch file if you dont want to commit it now
  - change to feature branch "new sensor"
  - merge latest dev into feature
  - dev now on feature branch
  - commit if done to feature branch
- if new sensor is done, merge into dev

if you are looking for a good windows/mac git gui, try sourcetree (Free) or tower (pay).
those support git flow and easy merging. i use sourcetree.

im not sure about the debug file, this should be tested.
`
echo debugfile merge=ours' >> .gitattributes
`
that way, it should not get created in teh master if the master dont has it, but as said, im not sure.

ascillato2, 2018-07-10:
Hi,

If your question is addressed, please, close this issue. Thanks +1

closed 2018-07-10 by GasTurbineMan