arendst/Tasmota #3157: Considering abandoning Arduino IDE & Adopting VSC Instead - but errors :(

meingraham, 2018-07-07:
I had finally gotten the Arduino IDE to build & upload TASMOTA. But it's finicky. Any time I reboot or change the board type (e.g., 8266 to 8285) it ends up erroring out (can't find Arduino.h). I end up having to blow away the IDE, reboot, reinstall the IDE, and then stuff works again. That's strange behavior that I have not figured out how to avoid. It's a pain.

I'd read Theos tips January 10 post re: making the switch to VSC w/ the PlatformIO integration. I'd avoided it because there's so much information on the Arduino IDE setup (SuperHOuse, DrZzs, etc.) that got me this far. But I'm fed up with the "instability" of my Arduino IDE environment. I'm finally ready to take the plunge.

But... I'm hitting some errors. If begins the build and creates the firmware.elf. It then raises a warning (? it's in yellow)
> '"esptool"' is not recognised as an internal or external command, operable program or batch file.

Followed by the following error:
> *** [.pioenvs\sonoff\firmware.bin] Error 1

I have VSC and I installed the PlatformIO extension. Basically , I followed the PlatformIO IDE setup in the TASMOTA wiki. Although there is a separate Esptool wiki entry for using that tool, the PlatfomIO installation does not reference any Esptool dependency. Do I need to install that? Should PlatformIO have "brought along" Esptool as part of it's installation? In any case, how do I resolve this issue?

Thanks.

Mike

reloxx13, 2018-07-07:
Atom+PlatformIO is very easy to use.

Sometime ago i wrote a guide for this, maybe you try it :)

https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build

ascillato2, 2018-07-10:
Hi,

If your question is addressed, please, close this issue. Thanks :+1:

meingraham, 2018-07-10:
Adrian,

The suggestion to use Atom does not address my question thus I have not closed this track. I was hoping to get some information directly related to VSC & PlatformIO first. I still have not been able to get that environment to function properly for me.

Mike

ascillato, 2018-07-10:
Hi. As that is not a Tasmota issue, may be you won't find many answers to this.

I think would be better to as at the platform.io repository. I'me sure you will have your answer there.

meingraham, 2018-07-10:
Adrian,

Yes, I am "corresponding" on other forums re: this question. There were two reasons why I'd posted here. 1. was Theo's January recommendation... and 2. It is compilation of TASMOTA that is generating the error. Perhaps pointing to a specific configuration of my TASMOTA "sketchbook" or some missing information from the TASMOTA wiki on setting up PlatformIO. So, it's not just the "esptool" error or a generic VSC & PlatformIO setup... but a use of those those w/in the scope of TASMOTA. Thus the post here.

Your thoughts on my reasoning?

Regards.

Mike

ascillato, 2018-07-10:
Understood. No problem at all. Was just an advise.

I use Arduino and also platform.io.

For Arduino I use 1.8.2 and I added Tasmota libraries (copy and paste as
the wiki) and esp8266 library 2.3.0. I use that when I will upload to a
device. I found the binaries are more stable.

For Platform.io I use with Atom. Mainly for Testing and programming. It has
a better IDE.

I never used VSC.

meingraham, 2018-07-13:
- Uninstalled VSC & extensions.
- Rebooted.
- Installed esptool.
- Installed VSC.
- Installed the PlatformIO VSC extension.
- Expanded the TASMOTA source code archive to my PlatformIO projects workspace.
- Edited platformio.ini (the one supplied with TASMOTA).
- Enabled and edited user_config_override.h.
- Compiled & linked successfully.

What was confusing was that esptool is "billed" as the upload tool. However, I was only attempting to compile the code (initially) - i.e., no upload requested. Nevertheless, esptool was required to successfully complete the compile and link.

I am now able to successfully "fully" compile (i.e., compile & link) and also compile and upload.

I'm wondering if the VSC installation wiki ought to be updated to explicitly include the requirement to perform the esptool installation?

Regards.

Mike

closed 2018-07-13 by meingraham