arendst/Tasmota #3158: SwitchMode/SwitchTopic - I guess I'm dense or outright stupid

meingraham, 2018-07-08:
I've read the docs, several posts, and watched DrZzs tutorial... and I **still** don't know if what I want to do is possible. And if it is, what the settings need to be.

I have a Sonoff 4CH Pro. I have only the "stock" hardware with TASMOTA flashed on it. I have no additional "external" input switches. The behavior I need is for the buttons on the device which normally control the relays to be "disconnected" from the relay altogether. If I press one of the four channel pushbuttons, I want the device to send an MQTT message stating that the button was pressed and what state the corresponding relay is in. BUt I don't want it to take an action on the relay. Then, once I receive the MQTT message, my openHAB rule logic will run and send back the appropriate MQTT command(s) to set the desired relay(s).

The use case I have is regarding control of ceiling fan speeds. I have five fans in the house and it seems that each manufactured implements the sped selection differently. Some speed switches are interlocking (at most two of the inputs are connected at any one time). Other fan implement the speed settings my connecting multiple inputs in varying permutations. When I replace the physical switch with a software switch, I need to have different logic for each type of fan. I don't want inadvertent user error where they select more than one relay at a time. And, yes, I know I can set up each 4CH differently (interlocking / self-locking) according to the type of fan it's controlling. But, because I want to use three of the relays for the fan speeds and the fourth relay for the fan light, I need to have the 4CH set to self-locking so I can have the light on and the fan on simultaneously. I will set up the interlocking via openHAB rules. Thus the need to receive a "request" from the switch and I control the fan speed relays (multiple simultaneous or interlocking according to fan manufacturer).

Thanks.

Frogmore42, 2018-07-08:
Note that the buttons know nothing of what state the relays are in.  You will have to keep track of that and/or set them the way you want them.  Depending on your latency requirements, you might even be able to query the device to find out.

ascillato2, 2018-07-10:
Hi,

If your question is addressed, please, close this issue. Thanks :+1:

meingraham, 2018-07-10:
Adrian,

Sure, I'll be sure to tidy up. I haven't had an opportunity to attempt the suggested configuration. I have ordered some new 4CH units and am going to give this feature a try on one of them to see if I have any further questions on how the feature works exactly. I'll close the issue once I successfully attempt using the setup.

Cheers!

Mike

meingraham, 2018-07-31:
I ended up using rules to keep the functions in sync whether the action was done on the physical switch itself or whether the action was performed through openHAB.

closed 2018-07-31 by meingraham