arendst/Tasmota #3160: Firmware upgrade issue

mskrohn, 2018-07-08:
I just OTA upgraded my Sonoff basic from firmware 5.14 to 6.1.0 and now it is stuck in a loop and I can't connect to it.  I first had to upgrade to the 6.1.0 sonoff-minimal.bin with no issues and then upgraded with the 6.1.0 sonoff.bin.  The upgrade completed "successfully" and after the reboot, the sonoff turns on with a solid green LED for about 40 seconds, then rapidly flashes 30 times, clicks with 2 more flashes and then repeats endlessly.  
If I try to go to the ip address I get This site can’t be reached 192.168.1.30 took too long to respond.
If I constantly refresh my routers connected devices page, the ip address does show up for about 10 seconds during the solid LED part of the cycle.
The button does cycle the relay properly but has no effect on the loop its stuck in. I also tried holding in the button and that has no effect either.

Do I need to hardwire reflash it or is the something simple I am not seeing?

PJFonseca, 2018-07-08:
Your answer is here: https://github.com/arendst/Sonoff-Tasmota/issues/3150

reloxx13, 2018-07-08:
> a solid green LED for about 40 seconds, then rapidly flashes 30 times, clicks with 2 more flashes and then repeats endlessly.

wifi connection failed, WIFICONFIG started, default WPS_MODE if not changed.

helmar74, 2018-07-10:
Also commented in [#3150](https://github.com/arendst/Sonoff-Tasmota/issues/3150)

Hi, 
I had the same issue. All sonoffs (S20, 433 MHz, 4CH Pro, SC, B1...) are working well with Stable Varsion 6.1.0 and german translation

...Except the sonoff Basic. It is possible to update to minimal version 6.1.0 via OTA, but Update to "normal" version isn't possible. Device was stuck.
That's how i was able to fix:
1. Used Development Version 6.1.0a (Stable version 6.1.0 isn't working for me)
2. Don't use german translation (if used, flashing via serial connections works, but no wifi connection is possible):
     Commented out the following in user_config.h:
     //#define MY_LANGUAGE            de-DE           // German in Germany
3. with this change I was able to update via serial connection
4. OTA Update to minimal version and then back to "normal" version isn't possible at all

Regards

Helmar

ascillato2, 2018-07-16:
Hi

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2