arendst/Tasmota #3164: DHT11 fractional temperature is ignored

hydrogen18, 2018-07-08:
**Describe the bug**

I bought some Sonoff Basic from AliExpress along with DHT11 sensors. I got Tasmota installed and a DHT11 installed. I saw temperature readings immediately, but they have only 1 C resolution while the device is capable of higher.

I tracked this down in the code to this line https://github.com/arendst/Sonoff-Tasmota/blob/v6.1.0/sonoff/xsns_06_dht.ino#L161
```
t = dht_data[2];
```
So the conversion consists of promoting a `uint8_t` to a `float` and calling it done. As far as I can tell the internet believes these devices don't output anything but zero for the last byte. For example:
http://sheepdogguides.com/arduino/ar3ne1humDHT11.htm

But here are my log lines from that file I see in the web console.

```
18:03:53 DHT: Received 53, 00, 1B, 05, 73 =? 73
18:03:55 DHT: Received 56, 00, 1B, 06, 77 =? 77
18:03:59 DHT: Received 58, 00, 1B, 09, 7C =? 7C
18:04:02 DHT: Received 5B, 00, 1C, 03, 7A =? 7A
18:04:05 DHT: Received 4F, 00, 1C, 04, 6F =? 6F
18:04:09 DHT: Received 45, 00, 1C, 06, 67 =? 67
```

So obviously I am seeing non-zero for the last byte. It is 0x0 - 0x9 in value. So it's fractional decimal. I refactored the code to be the following

```
      t = dht_data[2];
      for(uint8_t i = dht_data[3]; i != 0; i--){
        t+= 0.1f;    
      }
```

And the readings I get now have tenth of a C resolution and make sense as well.

The DHT11s I got are nothing special, just the cheapest thing on AliExpress as of a week or two ago.

Here are images of my setup

[image]
[image]

**If you will accept this change, I will submit a pull request**

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues and in the wiki
- [ x] Hardware used : Sonoff Basic R2 V1.0 2017-10-11 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
Use a DHT11 sensor, observe you only get 1C resolution

**Expected behavior**
Use a DHT11 sensor, get the 0.1C resolution the sensor provides

reloxx13, 2018-07-08:
read the wiki, read the user_config.h file, search for resolution.

stop wasting your time, drink a beer 🍺

hydrogen18, 2018-07-08:
All that does is become the "precision" that is passed into this function

https://www.nongnu.org/avr-libc/user-manual/group__avr__stdlib.html#ga060c998e77fb5fc0d3168b3ce8771d42

That number could be 100, it'd just show 100 zero's after the decimal. You cannot create the fractional component of a temperature sensor with a library function.

andrethomas, 2018-07-08:
I'm not sure fractional precision on a DHT11 is useful anyway - the value will just jump all over the place anyway... the DHT11 is about as stable as a donkey on steroids!

If you want precision rather use a BMPxxx/BMExxx sensor or even an LM75AD - they are fairly cheap and much more reliable in terms of accuracy and hysteria on temperature reporting.

andrethomas, 2018-07-08:
@hydrogen18 Did you do some output logging to check the stability of results when adding 0.1-degree resolution?

arendst, 2018-07-09:
@hydrogen18 nice write-up but...

- Your DHT11 has a resolution of +/- 2 degrees Centigrade which makes decimal values useless.
- You might have selected device AM2301 (=DHT22) using the same protocol to talk to the device and uses your "decimal" value out of the box.

hydrogen18, 2018-07-09:
@arendst I haven't read the spec sheet for the DHT22 but the code used to compute its value is  different. Here's the code for the DHT22

```
t = (((dht_data[2] & 0x7F) << 8 ) | dht_data[3]) * 0.1;
      if (dht_data[2] & 0x80) {
        t *= -1;
      }
```

This looks rather different, don't you agree? I'm not sure what they are trying to implement.

With respect to your comment about the sensor, I think you are confusing accuracy vs precision ( I don't know the Dutch words) . Here is a good example of the difference:

https://labwrite.ncsu.edu/Experimental%20Design/accuracyprecision.htm

I don't have any equipment other than a typical Type K thermocouple, so it'd be difficult for me to characterize one of these sensors. But my point is they can have very high precision while still having a poor accuracy of +/- 2 C.

andrethomas, 2018-07-09:
@hydrogen18 

```
t = (((dht_data[2] & 0x7F) << 8 ) | dht_data[3]) * 0.1; ** gets byte 2 and 3 to get 10th fractions as you would expect
      if (dht_data[2] & 0x80) { ** checks if the temperature is negative?
        t *= -1;
      }
```

Again, the results on the DHT11's I have appear to contain nothing in data[3] as quoted earlier up

[image]

andrethomas, 2018-07-09:
@arendst @hydrogen18 I guess since there is contrast in information available on the DHT11 insofar that some of them provide data in data[3] and others provide a 0 in data[3] (such as the ones I have tested)

Knowing this, adding a PR or manual edit to add code of @hydrogen18 would not imact negatively on sensors which provide a 00 on data[3] anyway, so it will either have 10th temperatures, or not have them depending on whether the sensor sends datain [3]

The two lines suggested by @hydrogen18  will not negatively impact so either add it manually or @hydrogen18  must make a PR for it - Then its fixed at least for those people who are lucky enough to have sensors that provide data in data[3]

Suggestion is therefore to add the following code as it has no negative impact on sensors that do not produce data in data[3] but may have befits for those that do.

```
for(uint8_t i = dht_data[3]; i != 0; i--){
        t+= 0.1f;    
      }

```

arendst, 2018-07-09:
Updated DHT driver.

Give it a try and let me know the results.

andrethomas, 2018-07-09:
@hydrogen18 

I think the original code was written with this one in mind:
https://osepp.com/wp-content/uploads/2015/04/DHT11-Technical-Data-Sheet-Translated-Version.pdf

In that, it only mentions 1degC
[image]

I think it may be possible that your DHT11 actually contains the same chipset as a DHT12, which has been referenced to provide 0.1 deg C resolution... e.g.

http://www.robototehnika.ru/file/DHT12.pdf

[image]

In either event, let us know what happens with the code change that was made.

hydrogen18, 2018-07-12:
Hi, I have not had time to test your changes yet.

@andrethomas I agree this is probably a DHT12 labeled as a DHT11. No real reason for a factory to bother manufacturing both.

ascillato2, 2018-07-16:
Hi,

If your question has been addressed, please close this issue. Thanks :+1:

closed 2018-07-21 by ascillato2