arendst/Tasmota #3166: Additional Switch on 4CH Pro with KNX strange behavior

ss83, 2018-07-08:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : Sonoff 4CH Pro
- [ ] Provide the output of command ``status 0`` :
```
22:36:29 RSL: STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:36:29 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff-knx.bin","RestartReason":"Software/System restart","Uptime":"0T03:11:03","StartupUTC":"2018-07-08T18:25:26","Sleep":0,"BootCount":117,"SaveCount":859,"SaveAddress":"F4000"}}
22:36:29 RSL: STATUS2 = {"StatusFWR":{"Version":"6.1.0a","BuildDateTime":"2018.07.07 22:36:01","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:36:29 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["WLAN",""],"TelePeriod":300,"SetOption":["0A008001","55818000"]}}
22:36:29 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":540,"Free":460,"Heap":11,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","1F802794","0C000000","35B6179E","00000000"]}}
22:36:29 RSL: STATUS5 = {"StatusNET":{"Hostname":"4CHPro","IPAddress":"192.168.0.123","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"50:03:95:96:DE:1F","Webserver":2,"WifiConfig":2}}
22:36:29 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 08 21:36:29 2018","Local":"Sun Jul 08 22:36:29 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:56","Sunset":"20:54"}}
22:36:29 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-07-08T22:36:29","Switch5":"OFF"}}
22:36:29 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-07-08T22:36:29","Uptime":"0T03:11:03","Vcc":3.174,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"WLAN","RSSI":78,"APMac":"76:4A:21:4B:BB:57"}}}
```

Hi,
I have some strange behavior that I don't understand. Have been searching the last two days without success.

The hardware is a Sonoff 4CH Pro that I would like to use with Tasmota KNX. I flashed the latest Firmware 6.1.0. The normal communcation over knx works good, atm I just use relays 1-3.
You can see on the pictures below that I have configured different group adresses for switching the Relais and for sending out the state. This works good and reliable.

Now I would like to attached some sort of switch / reed contact to the 4CH Pro. I only Need the rising flank to send out a "1" on a specific Group adress. So i configured the GPIO2 as Switch5 and set up a rule:
`21:55:39 CMD: rule1
21:55:39 RSL: RESULT = {"Rule1":"ON","Once":"OFF","Free":471,"Rules":"on Switch5#State=1 do KnxTx_Val1 1 endon"}`

When I close the contact on the Switch GPIO2 I get the following:
`22:02:13 RUL: SWITCH5#STATE=1 performs "KnxTx_Val1 1"
22:02:13 KNX: KNX TX 1 = 1.00 gesendet an 6.4.0
22:02:13 RSL: RESULT = {"KnxTx_Val1":"1.00"}`
That seems to be correct and on the KNX-Bus i get the correct value to 6/4/0.

But now the strange Thing happens: When I open the contact on GPIO2 I get:
`22:02:36 KNX: Ausgang 1 = 0 gesendet an 1.4.1
22:02:36 RSL: RESULT = {"POWER1":"OFF"}
22:02:36 RSL: POWER1 = OFF`
Why does the Relais1 then Switch off? What can I check to avoid this?

Pictures of my config:
[image]

[image]

Regards
Sven

ascillato, 2018-07-09:
Hi,

If no behaviour is defined when _switch5 = 0_ it will change the state of the default relay (in your case is one because there is no relay5 defined)

Try to change your rule to:

`rule on Switch5#State=1 do KnxTx_Val1 1 endon on Switch5#State=0 do delay endon`

ss83, 2018-07-09:
Hi,

thanks, the missing event for witch = 0. Adding the delay solved the problem.

Regards
Sven

ss83, 2018-07-09:
Resolved#3166

closed 2018-07-09 by ss83