arendst/Tasmota #3167: Firmware issue

vaidyasr, 2018-07-09:
I have upgraded to the new firmware for my Sonoff basic, and the MQTT state topic got changed from
state_topic: "stat/sonoff/POWER" to state_topic: "stat/sonoff/POWER1" with 1 at the end. Is that normal?. or do I need to configure anything?

Frogmore42, 2018-07-09:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
SetOption26

You have to provide all of the info listed in the template for help with the other issue.

ascillato2, 2018-07-10:
Seems that  your issue is solved. If not, please reopen it. Thanks

closed 2018-07-10 by ascillato2

vaidyasr, 2018-07-11:
The following is the config.
> - platform: mqtt
>   name: "Home Theater Switch"
>   state_topic: "stat/hometheater/POWER1"
>   command_topic: "cmnd/hometheater/POWER"
>   availability_topic: "tele/hometheater/LWT"
>   qos: 1
>   payload_on: "ON"
>   payload_off: "OFF"
>   payload_available: "Online"
>   payload_not_available: "Offline"
>   retain: true

The SonOff randomly switched on.

Frogmore42, 2018-07-11:
You have retain set to true, that will cause strange things to happen.  Search the issues for retain and will see all the people who have run into that.  You need to set it to false and clear out all the retained messages from your broker.  The other issues have more details.