arendst/Tasmota #3168: SerialSend as hex

KPK75, 2018-07-09:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Yes, support for LCTech modules

**Describe the solution you'd like**
Ability to send serial commands via MQTT like SerialSend4 0xA00101A2

bastelfix, 2018-07-09:
check this: https://community.openhab.org/t/looking-for-a-mqtt-to-ir-device/46455/7

Jason2866, 2018-07-09:
Hello @KPK75 
Only this is possible
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#serial-bridge
@arendst did the design decision not implementing sending Serial Raw only printable character.
See #2190

ascillato, 2018-07-17:
Hi @arendst 

Is it possible to be added a command SerialSend4 to send Hex values?

ascillato2, 2018-07-29:
Added by Theo.

Closing issue. 

Thanks :+1:

closed 2018-07-29 by ascillato2