arendst/Tasmota #3170: big problems calling the configuration page

dr-apple, 2018-07-09:
I have big problems calling the configuration page.

Either the site just breaks or it is empty or it is displayed but you can not see any settings and all dropdown fields are empty and you can not select anything.

No matter with which browser I try it always the same ;-(

curzon01, 2018-07-09:
Please do not pass the requested information when posting an issue. I assume what could cause the problem but guessworking cost a lot of time.
Please post at minimum Status 0 info or if not possible if you use which release binary or self-compiled, how compiled (Arduino, Atom...) which version you use and which lib you select

reloxx13, 2018-07-09:
http://yourdeviceip/cm?cmnd=status%200

if you cannot reach the console web page.

ascillato2, 2018-07-16:
Hi

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

closed 2018-07-16 by ascillato2