arendst/Tasmota #3175: Is possible add virtual buttons?

yonaesp, 2018-07-09:
I use the sonoff dual for my blinds, I want know if is possible add virtual buttons for the same relay with different pulse time for set the position based on the upload/down time.

Regards.

ascillato, 2018-07-10:
There is not virtual buttons. But you can use rules for making that behaviour. Please read the wiki. Thanks

yonaesp, 2018-07-10:
@ascillato I do not want you to give me the work done, but could you give me a little more guidance on what to look for? thanks

yonaesp, 2018-07-10:
@ascillato I think I will not achieve it with my simple knowledge that does not go beyond installing the firmware and copying commands in the console.

There is a special [firmware for blinds](https://github.com/kabongsteve/Sonoff-Tasmota) but I think it does not do what I want or maybe the instructions are not very clear. 

I do not understand why you can not add several virtual buttons with different pulsetime for the same switch, this would be a good solution for many people who want to use the sonoff in blinds

ascillato, 2018-07-10:
Let's see. What do you mean about virtual button?

A button in the webpage?
A button in a tablet (dashboard) ?
A command called by you HomeAutomation Software?

yonaesp, 2018-07-10:
@ascillato the dual sonoff, has 2 buttons and two switches

I would like to add more buttons for the same switch, but with different pulsetimes.

ascillato, 2018-07-10:
Ok, so you have a modified Sonoff Dual with 2 buttons and 2 switches.

You want that

* Button1 when pushed turn relay 1 on for x seconds
* Button2 when pushed turn relay 2 on for y seconds
* Switch1 when pushed turn relay 1 on for z seconds
* Switch2 when pushed turn relay 2 on for w seconds

Is that right?

yonaesp, 2018-07-10:
@ascillato it's perfect :)

ascillato, 2018-07-10:
Ok, give sometime. I'm going back to you.

ascillato, 2018-07-10:
Let's make a rule for that behaviour (this is just configuration, no programming needed)

**1 - Module Configuration:**

So, assuming that:

* You are using **Generic Module** in configuration
* And that physically you have **2 pushbuttons and 2 wall switches** attached to the GPIOs of a sonoff Dual.

So, let's configure all inputs as switches, so we will have:

* switch1: to the GPIO of your pushbutton1 (when push it makes GPIO go GND)
* switch2: to the GPIO of your pushbutton2 (when push it makes GPIO go GND)
* switch3: to the GPIO of your wall switch1
* switch4: to the GPIO of your wall switch2
* And of course you will have Relay1, Relay2 and Led1

**2- Inputs Behaviour:**

Behaviour desired: When a pushbutton is pushed or a switch is toggled, execute the action of point 3.

Go to the console and type:

```
Switchmode1 3
Switchmode2 3
Switchmode3 7
Switchmode4 7
```

**3- Outputs Behaviour (done by rules):** 

Behaviour desired:

* Switch1 go up 10 seconds
* Switch2 go down 10 seconds
* Switch3 go up 1 second
* Switch4 go down 1 second

Go to the console and type:

```
rule 1
rule on Switch1#State do event goup=10 endon on Switch2#State do event godown=10 endon on Switch3#State do event goup=1 endon on Switch4#State do event godown=1 endon on event#goup do backlog power2 0; pulsetime1 %value%; power1 1 endon on event#godown do backlog power1 0; pulsetime2 %value%; power2 1 endon
```

**4- Extras**

You also can control this by MQTT using
```
cmnd/Sonoff/event goup=1
cmnd/Sonoff/event godown=1
```

This is just an idea, I do not have now a hardware like yours to test it. May need some adjustments to these configurations.

After your comments, we can add this example to the wiki. Thanks for sharing your idea. :+1:

yonaesp, 2018-07-11:
@ascillato 
Thank you very much for your incredible help and effort.

Tell me if I understood correctly. Do I need to connect 4 phase cables to the sonoff to turn it into something like a 4ch sonoff?

When you say that I need two physical pulse switches and two physical switches, I think I can not directly connect to the shutter motor, which has only two phase cables, up and down.

Or do you mean 4 physical switches by way of example but meaning that they are inside the sonoff?
I apologize that I can not understand completely, it's something a bit complex haha

ascillato, 2018-07-11:
Please, share a photo of what is the hardware you actually have with the buttons you said.

yonaesp, 2018-07-11:
@ascillato 
Button 1 is associated with relay one (when I said relay or switch I always referred to the outputs of the sonoff, maybe I explained wrong, I do not speak English very much, and the translator less)

I will explain again, to see if this is better understood.

I have sonoff dual R2, has only two current outputs.
Toggle 1 > activates the current output 1
Toggle 2 > activates the current output 2

this is what I want:
Toggle 1 > activates the current output 1
Toggle 2 > activates the current output 2
Toggle 3 > activates the current output 1
Toggle 4 > activates the current output 2
each toggle would have a different configuration, for example, 1 and 2 for 15 pulsetime for seconds, 3 and 4 for 2 seconds.

I thought you wanted to say this, but I think I did not understand you

the engine that I use has 4 wires, ground, neutral and two phases, (one phase for rise and another phase for descent.

For example, I can connect two bulbs and they work independently, and I would like to have 4 toggles, two for each bulb

ascillato, 2018-07-11:
So, the buttons you are refering are the ones from the webpage?

yonaesp, 2018-07-11:
@ascillato 
Exactly, also those that appear on my devices when I activate the emulation hue, since I would like to control also from the other devices like google assistant and alexa, for this reason I want more "buttons", but for the same outputs of energy

yonaesp, 2018-07-11:
I understand that this could be done simply with a 4ch sonoff, connecting 2 relays to rise and two relays to descent.

I wanted to be able to do it with the dual, since that could also for example establish "buttons" like:
Button 1: Raise the blind to position 1 (activate relay 1 for X seconds)
Button 2: Raise the shutter to position 1 (activate relay 1 for Z seconds)
Button 3: Raise the shutter to position 1 (activate relay 1 for Y seconds)

ascillato, 2018-07-11:
Ok, Sorry, I understood another thing.

So forget everything about the rules and the example I did.

Adding buttons to the webpage is not possible because the size of the program flash and the RAM usage. So, sorry, no buttons.

If you want to add that behaviour with different pulsetime you can use a home automation system and pass those parameters to Tasmota to behave like that.

Tasmota is intended to be a MQTT capable device for Home Automation integration. It is not intended to be a stand alone device.

You need a MQTT server and also a home automation software like Home Assistant, OpenHab, Domoticz, NodeRed, etc.

Sorry

yonaesp, 2018-07-11:
oh... I thought that if it would be possible :/, thank you for your help :)

ascillato, 2018-07-11:
Thanks to you for your patience :+1:

ascillato, 2018-07-11:
So,

Let's close this issue. Please, if you think that something else is needed for this topic, reopen this issue.

Thanks.

closed 2018-07-11 by ascillato2

yonaesp, 2018-07-11:
this will be what I will do then, I put it in case someone ends here because he wants the same thing.

Sonoff dual 4ch pro (pro since it has hardware interlock, which gives a security bonus in case the configuration is erased), since we are going to use it in an engine with the option to go up and down and we can not drive both to the time.

By having 4 energy outputs we would do the following:
Output 1, we would call it upload and it would activate for 15 seconds
Output 2, we would call it descent and it would activate for 15 seconds
Output 3, we would call it low a little and it would activate for 3 seconds
Output 4, we would call it up a little and it would activate for 3 seconds

Regards.

ascillato, 2018-07-11:
Like a lot your approach!!!

Thanks again for share :+1:

CaptainHindsigh, 2025-05-15:
[image]