arendst/Tasmota #3176: Issue of Sonoff Touch (2 gang) as Roller Shutter

Sunrise17, 2018-07-10:
@davidelang , @arendst , i have uploaded 6.1.0 to Sonoff Touch (2 gang model) to use it as Roller Shutter switch. I have adjusted Setoption14 1, Setoption 13 1 and when i use it with following homebridge plugin below, it works well during open/close switch on main menu and even over Siri (to ask percentage). But on Slider, there is some weird issue to update current state of the switch as seen in the video following below, i will be glad if you can share your comments.

[homebridge-tasmota-mqtt-motor Plugin](https://github.com/ageorgios/homebridge-tasmota-mqtt-motor)

[Video Link](https://www.youtube.com/watch?v=GXmC4nBuQmY)

chriskmn, 2018-07-10:
I have an issue with SetOption14 1: turning off an switched-off port turns off also the other port if it is active. Can that be your problem also ?

opendore, 2018-07-11:
Also me, same issue...any idea?

stefanbode, 2018-07-16:
If you use it for shutters you may better go with this fork: https://github.com/stefanbode/Sonoff-Tasmota/wiki/Shutter-blinds-working-with-two-relays. To work with pulsetime is more than nasty.

ascillato2, 2018-08-25:
Hi,

Continuing this issue at: #3137

closed 2018-08-25 by ascillato2