arendst/Tasmota #3178: Is my Sonoff_basic Faulty?

SSkull995, 2018-07-10:
Hi everyone, thanks in advance for the help and the attention. I try to explain my issue: I have a Sonoff basic which I successfully flashed tasmota 6.1.0 on using platformio. I have encountered an issue as soon as I have connected it to the AC power: the led doesn't blink. When the FTDI dives the power everything works fine. Moreover, I have tried to connect just the RX TX and GND to the FTDI and the Sonoff to AC, to see if the serial monitor had caught something, nothing. It seems that the power doesn't arrive to the module and doesn't power up the unit at all. I have a pi zero with HAP-NodeJS and I can actually control the sonoff if the power arrives from the FTDI, otherwise nothing. 
I read a lot on the Wiki, I changed the ESP to 1.3.0 and I tried with an old firmware (5.13.1), nothing changes. Do you know which the problem could be? I am going crazy. Thanks!

Frogmore42, 2018-07-10:
Sounds like your device's power supply is not working.  Depending on how much you know about electronics and what equipment you have it might be cheaper/safer to just get a new one.

If you have a meter and know how to use it safely, you can measure between VCC and ground and see if you see 3V3 or not.  If not, then check if you see 5V on the input to the regulator.  If you see 5V but not 3V3, the regulator is bad.  If you don't even see 5V, probably best to get a new Sonoff.  They are pretty reliable, but I have had a couple of failures.

SSkull995, 2018-07-10:
Hi, unfortunately I don’t have a meter in this house. However I was wondering, maybe I can connect 3.3 and AC? Worst situation=fried chip right?

SSkull995, 2018-07-10:
Yeah no of course I would avoid anything stupid. The only frustrating thing is that I used this cable for my 3D printer ahahahah, heck of a Sonoff

SSkull995, 2018-07-10:
Ah I have another issue with a Sonoff S20. If I connect it to the FTDI nothing flashes, it neither enters in boot mode if keep the button pressed nothing. At the beginning it switched on when connected to AC, now it doesn’t do that anymore. If I use the 5v, the blue led lights up, otherwise nothing at all. This one is bad too?

Frogmore42, 2018-07-10:
Since you have no tools to see what is really going on, your options are limited.  Since you have two devices that now appear to be broken, but once we're not, it seems your best option is to purchase a cheap volt meter and learn how to use it.

Some FTDI adapters output 5V on VCC even when set to 3V3 (which just changes the logic levels).  Mine, which is a real one does this.  It is marked 3V3, but the VCC line is connected directly to VBUS on the USB port, which is 5V.  The datasheet does indicate this.  Some other manufacturers use the 3V3 output from the FTDI part, but it is not rated to provide enough current for an ESP8266.  It mostly works for some people though.

Anyway, without a meter you and everyone else will be guessing.  That might get to a solution, but it probably won't be quick.

After breaking two of anything, I always take a long hard look at what I have been doing to see if it might have been the cause or if it really was bad quality parts or just really bad luck.

SSkull995, 2018-07-10:
Actually I am quite sure about the 5v output of the FTDI, if I chance the jumper to 3.3 it doesn’t light up the led so something changes. Anyway, I ordered 2 new Sonos, just to see through this mess. Honestly, I don’t know what’s the problem with the s20 and neither with the basic one. Perhaps the s20 got damaged because of wrong FTDI voltage, but I am sure that the basic one was not. It works on the bench with the adapter, not connected to the ac directly

Frogmore42, 2018-07-11:
I would also suggest a meter and spend some time learning to use it.  It will get expensive pretty fast if you destroy the next two.  You don't need an expensive meter to tell if there is voltage there or not.

SSkull995, 2018-07-11:
I have one, I know how to do things, I just don't understand why I don't get serial from a device that connected to the adapter work with my iPhone, while connected to the ac is like dead

SSkull995, 2018-07-11:
I will try to find a friend to borrow from it from

Frogmore42, 2018-07-11:
The most likely reason is the AC to DC power supply on the Sonoff is not functioning correctly.  The meter will help confirm that.  There are a few reasons why this might be.  It might even be as simple as the fuse is blown.  The meter is what you need in order to do basic failure isolation, ie figure out which part of the circuit is not functioning.  If it is the fuse that is blown, I would want to know what caused it to blow, since it is only supposed to do that to prevent a greater fault.

ascillato2, 2018-07-16:
Hi,

If your question has been addressed, please close this issue. Thanks.

closed 2018-07-17 by ascillato2

SSkull995, 2018-07-17:
Bro, why did you close it? I just received my Sonoff today. Anyway, I figured out that my Sonoff_basic actually works connected to the AC. I tried to flash my s20, but it does not work, it seems there is not enough current during flashing. If i just connect my ftdi adapter, the sonoff does not switch up. Should I change it?

Frogmore42, 2018-07-17:
The FTDI part is not rated to supply enough current for an esp8266.  Some people have successfully flashed with it, others have not been so lucky.  I have not heard of people having issues with CH340 ones.  I power mine from the USB supply with a regulator to bring it down to 3V3.