arendst/Tasmota #3179: Problem with WIFI Extender

spippolone, 2018-07-10:
Hi guys,
I have the Fiber Router, and 2 Wi-Fi extender. If i configure the sonoff with the Fiber Router all work fine, but if i connect the sonoff to a Extender one or Extender 2 ( all are in a same net 192.168.0.X ) I can't found the sonoff after reboot. 
I press 4 times S00 switch I connect my pc to sonoff net, in 194.168.4.1, I configure it in a my net by net search, then after reboot I saw the sonoff only if i work in a router fiber area, but no in the Externder area.
WHY this ? 
I have the last tasmota firmware 
Thank you

ascillato, 2018-07-10:
Hi, I found that the Esp have problems with some wifi exterders. Try to connect directly to your router

ascillato, 2018-07-11:
I don't know why it has problems with some network configurations. But it is like that. In my case, my laptop was able to connect to an access point without any issue but Tasmotas and my cellphone could not. So, I changed my network to have separated routers instead of the same network repeated. Also I lower the amount of wifi devices wiring everything that could be connected directly by wire (TV, etc)

ascillato, 2018-07-11:
Similar issues (Added for reference)
#2562
#2864

spippolone, 2018-07-11:
Ascillato thanks four help my situation is different...  I have main fiber router, with your WIFI spot, and an Asus Access Point with another Spot connected to main router by LAN cable. My PC is FIXED connected by LAN cable to main router, NOT wifi connected.. and then the sonoff in a main router spot work fine in the Asus Access Point Spot NOT working..  I will go to read your linked issues .. but ... you work with a PC conncted in "spot 1" and sonoff connected in "spot 2"  .. my PC is cabled.

ascillato, 2018-07-11:
Yes, exactly. Sometimes the ESP8266 Libraries has problems with some wifi extenders or access points. Can't be solved inside Tasmota Software. May be in the future, the esp8266 core libraries for Arduino, got this solved.

spippolone, 2018-07-12:
Fine, I understand-...  I hope that the problem will go to solve, more people use some extender.. in the house.