arendst/Tasmota #3180: v6.1.0 Sonoff-ru.bin WiFi autoconfig doesn't start

i3laze, 2018-07-10:
Didn't test EN firmware, but tried erasing/flashing fresh new Sonoff Basic three times.

Quickly pressing button 4 times - LED starts blinking, but Sonoff-ID SSID doesn't appear in WiFi networks list.

Just flashed 5.14 - 4 presses and autoconfig SSID is right here. Please investigate.

andrethomas, 2018-07-10:
Is there any output on the serial port?

i3laze, 2018-07-10:
I'll try to reproduce on another unit. Hold on for an hour.

andrethomas, 2018-07-10:
The content in the serial output is questionable... should not have all those special characters.

Have you tried configuring wifi from the serial console, enter commands

ssid1 "mywifiapname"
password1 "mywifipassword"

andrethomas, 2018-07-10:
Correction, dont include the "'s

Just

ssid1 mywifiapname
password1 wifipassword

i3laze, 2018-07-13:
Nah, I've deployed v5.14 back in production.
Let's wait for anyone else to confirm the issue.

i3laze, 2018-07-18:
I guess it was solved in v6.1.1:
https://github.com/arendst/Sonoff-Tasmota/issues/3177
Will give a try today.

ashfaaaa, 2018-07-20:
I have the problem on v6.1.1  I just flashed tasmota and im  unable to connect with button 3 press and button 4 press ( led blinks) no ap found when i search for it.

ascillato2, 2018-07-21:
Please, see #3224

If your issue is not solved, please reopen it. Thanks

closed 2018-07-21 by ascillato2