arendst/Tasmota #3181: Backup/Restore in Plain-Text

Luzifer, 2018-07-10:
In order to be able to inject a customized configuration using an automation like Ansible or any other system it would be great to have another chance to create backups / import restores in another format (JSON?) than the default binary (memcopy?) format.

That way a JSON file could be generated using the automation tool and afterwards be pushed to the Sonoff device to update all settings at once.

arendst, 2018-07-10:
That would take a massive decoding of JSON to hex values within the device taking up too much code space and RAM.

Luzifer, 2018-07-10:
Maybe a more simple text format? Like an ENV file with `KEY=VALUE` combinations?

andyfitter, 2018-07-10:
Another option would be an ‘offline’ converter, one that takes in Json and spits out the default binary format?

Downside would be that it would need to be kept in sync with the main codebase but that isn’t impossible. I’m assuming all backups have embedded version numbers.   Maybe it could even use the Tasmota headers in some form to define the binary format.

Luzifer, 2018-07-11:
For a proper configuration management this converter would need to be able to convert into both directions: Reading the binary format and writing it (you don't want to push a new config if it's equal to the old one)…

Also a side project will get out of sync with the main codebase… I've seen this too many times… 🙁

That's why I'd prefer an on-board solution…

andyfitter, 2018-07-11:
I agree with all your points (That’s why I mentioned some of them in my post :-)

As the suggestion has been dismissed I was suggesting it as potentially the only other solution.

patbeirne, 2018-07-14:
I started a personal project to solve this problem. It does indeed use its own .h file to keep a list of MQTT fields to fetch.

The main engine is here if anyone wants to try it. Just to see if this is feasible.
http://alt2.pbeirne.com/api/arduino/sonoffBackup_engine.py
http://alt2.pbeirne.com/api/arduino/sonoffBackup_h.py
Put them both in the same directory, modify the bottom of the _engine.py file, and run
   "python3 ./sonoffBackup_engine.py"

I will be wrapping a GUI front end, and a command line front end on this. 
And an SQLITE and plain-text backend on it as well. 

Please let me know if this is interesting, whether I should continue, and whether you want "restore" functionality.

Pat
patb@pbeirne.com

tobias-richter, 2020-01-07:
I came across this issue some days ago and I implemented an ansible role for sending commands to tasmota devices. Sadly this only helps people using tasmota with the webserver.

See: https://galaxy.ansible.com/tobias_richter/tasmota

Maybe this helps someone here :)