arendst/Tasmota #3185: Sonoff basic random switch on by its own

vaidyasr, 2018-07-11:
After new firmware upgrade, my Sonoff basic is switched on by itself automatically. Not sure why. The following is the config.

> platform: mqtt
> name: "Home Theater Switch"
> state_topic: "stat/hometheater/POWER1"
> command_topic: "cmnd/hometheater/POWER"
> availability_topic: "tele/hometheater/LWT"
> qos: 1
> payload_on: "ON"
> payload_off: "OFF"
> payload_available: "Online"
> payload_not_available: "Offline"
> retain: true

ascillato, 2018-07-11:
Hi. Your config seems ok.

How often is this turn on occurs? 

For your console output it is received the command from a mqtt server. 

Are you using Mosquitto or the embedded mqtt server of HA?

ascillato, 2018-07-11:
Please, Try retain false in you HA config

vaidyasr, 2018-07-11:
The switch on is random and am using HA. Before upgrade the retain flag was true and it never switched on automatically. After upgrade, I have changed to false, it got switched on automatically, so I thought because of retain flag and I have changed back to true.

vaidyasr, 2018-07-11:
Changed to false and will wait for a day.

Frogmore42, 2018-07-11:
Once you have set retain to true, you will have a retained message stored in your broker.  Restarting the broker does not clear the retained message.  Depending on the state stored in the retained message you might get undesired results of your device reconnects to the broker.

Boosik, 2018-07-11:
I had exactly the same problem with Sonoff Touch when using retain: true in HA. Setting retain to false fixed all issues for me.

vaidyasr, 2018-07-12:
Still the same issue, it keep on switching, on its own even after changed to false.

vaidyasr, 2018-07-12:
The 'Restart Reason' output from 'status 0' says Exception, is that something related with this?

Frogmore42, 2018-07-12:
You HAVE to DELETE any retained messages or they will still be retained (which is the design intent of a retained message 😀).

The exception is generally a problem, and that causes the device to restart.  When the device restarts, it connects to the MQTT broker.  It subscribes to the cmnd topic.  If there is a retained message, it gets it.  If the message says to Toggle, that is what the device will do, even if you have the device configured to remember the last state.

If you have configured the device to start at a particular state, say off, if it restarts with the load on, when it restarts, it will turn the load off.

There are a couple of ways to delete retained messages.  Search the issues and you will find the commands to use.  Do that first and then provide the log showing the issue if it occurs again.

vaidyasr, 2018-07-12:
I just deleted the mosquitto.db and restarted the mosquitto service fixed the issue.

ascillato, 2018-07-12:
Great!!!!!

If your issue is solved, please close it. Thanks. :+1:

closed 2018-07-12 by vaidyasr