arendst/Tasmota #3186: RF Relay switch on via MQTT

vaidyasr, 2018-07-11:
Am using Sonoff RF bridge with latest Tasmato firmware and I have a 4 channel relay which was paired with Sonoff RF bridge. Is there a way to switch on/off the RF relay using MQTT?

Frogmore42, 2018-07-11:
Read the wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
Rfcode and Rfkey is the place to look.

Any command can be sent via MQTT

vaidyasr, 2018-07-11:
When I pressed the button in Sonoff RF bridge webpage, there is no command topic triggered. How do I trigger via MQTT without knowing the command_topic, please help.

Jason2866, 2018-07-11:
RfKey<x>  5       Show default or learnt RF data

vaidyasr, 2018-07-12:
A command topic in MQTT is required for MQTT switch to trigger power on/off.

vaidyasr, 2018-07-12:
I have a 4 channel RF relay which was learned by the bridge and I can able to switch on/off from the web interface. I would like to do the same via MQTT so that I can control from homeassistant interface.

Frogmore42, 2018-07-12:
Rfkey is the command.  The exact syntax for the topic will depend on what your device subscribes too.
Have you sent any command via MQTT before?
I don't have a RF device so I don't know which rf switch you have configured.  Let's say it is the first one.  The command would then be:
RfKey1
Note there is no space between rfkey and 1

This page has all the commands and sample syntax for sending them
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

vaidyasr, 2018-07-12:
Rfkey1 is the command from RF Bridge console, how do I invoke via MQTT?

Frogmore42, 2018-07-12:
From the wiki page:
•Sending a command:

tasmota/sonoff-mylight/cmnd/Power1 ← "TOGGLE"
   ↳ // Power for relay 1 is toggled
   ↳ tasmota/sonoff-mylight/stat/RESULT → {"POWER1":"ON"}
   ↳ tasmota/sonoff-mylight/stat/POWER1 → ON

We've send the toggle command and received the new state confirmation.

To use it for your device:
Replace tasmota and sonoff-mylight with the names that your device uses.
Replace Power1 with RfKey1

Have you ever been able to send an MQTT command before?

vaidyasr, 2018-07-14:
I am able to control via MQTT and then I implemented in HA config. But the switch state is turned off after turned on. The config is

> - platform: mqtt
>   name: "Rf Switch4"
>   state_topic: "stat/sonoff_bridge/RESULT"
>   command_topic: "cmnd/sonoff_bridge/RfKey4"
>   qos: 1
>   optimistic: false
>   payload_on: "Learned sent"
>   payload_off: "0"
>   retain: true

Because of that I need to again switch on on the HA to switch off the RF switch

Frogmore42, 2018-07-14:
You have retain set to true.  Search in the issues for how to turn that off properly.  It will cause you headaches.

vaidyasr, 2018-07-14:
retain: false doesn't change the behavior. When I press the switch on the HA frontend, the switch button on the screen animates from off-> on and immediately moved back to off. How do I make it on, i.e. make the switch on the frontend on after pressing it.
I don't know whether I am explaining clearly.

Frogmore42, 2018-07-14:
You can't just turn off retain.  You also have to delete the retained message.  Retained messages are tricky and cause many problems for people.  There might be something else going on too, but a retained message is a likely cause of your issue.

vaidyasr, 2018-07-15:
Thank you, will get the help from homeassistant community.

closed 2018-07-15 by vaidyasr

bachoo786, 2023-02-28:
> Thank you, will get the help from homeassistant community.

hey @vaidyasr did you sort this out? I am having the same issue and its driving me mad.