arendst/Tasmota #3191: Sonoff POW R2 randomly stops working

c3reeman, 2018-07-12:
Ive got a sonoff pow r2 running tasmota 5.14 with an 85% WiFi signal it’s running a cellar sump pump I’ve got it configured in home assistant to give me a pump running or not running indication based on the power pulled when it does run ( pump turned on and off from float switch, pow R2 always on ) it’s been working great it’s enabled me to program In alarms for if the pump runs continually or the pump doesn’t run over a period of time etc

The problem is this morning I noted the pump had run but no power amps or anything was reported back to home assistant also noted that nothing was displayed on the tasmota dashboard so it’s not a tasmota mqtt home assistant fault, The fault was quickly rectified by a tasmota reset... just looking for thoughts on what may have happend and what I could do to avoid it happening again

arendst, 2018-07-12:
Fix some Pow R2 and S31 checksum errors (#1907)

Fixed in latest release v6.1.0

c3reeman, 2018-07-12:
My bad I thaught I was up to date, thankyou for your prompt response, great work

closed 2018-07-12 by c3reeman