arendst/Tasmota #3192: Firmware upgrade went bad - Reset not working

aymeric106, 2018-07-12:
Hello,

I had a SONOFF basic, FW ver 5.12 
Named SONOFF_TABLE_LIGHT, connected to MYNETWORK wifi

To upgrade via the web interface via file upload I thought the proper way was
actual FW (5.12) -> 5.14 minimal -> 5.14 -> latest minimal -> latest

I only did upgrade 5.12 to 5.14 minimal and it seems to complete (successful on the web interface)
But, after reboot, it does not connect to MYNETWORK anymore.
The led is flashing.
An SSID named SONOFF_TABLE_LIGHT has appeared but I can't connect (maybe because of the underscore)

Anyway, it I tried to reset by maintening the button for 40 secs but it doesn't work either. The led stops blinking, after 40 sec, it seems to reboot but still presents an unconnectable SSID SONOFF_TABLE_LIGHT.

I tried also to push the button 3 or 4 times like her https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage but nothing seems to happen.

How can I recover this ?

Thank you

Aymeric

maciboy, 2018-07-12:
Hi,
are you sure that your Basic is not running 6.1.0 now?
Since your issue sounds similar to the one we are having with 6.1.0 over here: https://github.com/arendst/Sonoff-Tasmota/issues/3177

aymeric106, 2018-07-12:
I'm pretty sure I flashed the 5.14 minimal from here

https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.14.0/sonoff-minimal.bin

But anyway, the seems to be the same issue you mentionned. Back to serial flashing I guess ...

Thank you

maciboy, 2018-07-12:
Strange that 5.14 seems to have a similar issue. For me it was quite stable and I'm thinking about downgrading to it from 6.1

kacho724, 2018-07-13:
I encountered the same issue on my Basic. The way that I fixed mine was to flash with serial connection. Below are steps what I did:
-Connect Basic by using serial (FTDI)
-Build and Upload 5.12 minimal in platformio
-Build and Upload 5.12 normal in platformio

After that my Basic is back to normal.

aymeric106, 2018-07-13:
I spent a few hours on this yesterday ...

**First, I confirm the issue,for me, it's when flashing 5.14 minimal over 5.12 that the Sonoff goes into the led flashing  mode with no connection** (even when flashing with the FTDI)

Then I tried various thing because after going back to 5.12 via FTDI, it seems to work then the web interface become slow, then the connection is lost. You can reboot, but again, the connection is dying.
I tried many things:
-send reset 1 in the command page,
-40 sec on the button
-change CFG_HOLDER in user_config.h and flash twice like explained here -> https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#cfg_holder-explained

But still the sonoff is not reliable now. This morning it was not responding to ping. Rebooted it and was still working 2 hours after  (go figure... never had any issue before I tried to upgrade)

The only thing I didn't do is to fully format the memory chip when compiling and flashing the firmware (because I don't know how to do it with the Arduino software I use https://www.arduino.cc/en/main/software)

aymeric106, 2018-07-13:
Just to be sure, I rebuild the firmware with 2.3.0 core instead of 2.4.0 and retry the whole process.
(I flashed all my sonoff basics with 2.4.0 core when I bought them)
Still the same issue when upgrading from 5.12 to 5.14_minimal with the 2.3.0 core.
But, with the 2.3.0 core, my sonoff basics are now **way** more responsive to MQTT request.
The only good thing I guess.

ascillato2, 2018-07-16:
Hi

If your issue is solved, please close it. Thanks :+1:

closed 2018-07-17 by ascillato2