arendst/Tasmota #3193: Sonoff POW max limit of total-kwh extended?

Loreman01, 2018-07-12:
Hey guys,

after reaching the total-kwh of 43 the value seems to be resetted to 0.

Is there an max limit of kwh that can be counted? Didn't found a variable to increase the limit in the settings. Am I blind?

arendst, 2018-07-12:
Pls provide all requested info before we can help you.

Loreman01, 2018-07-12:
I have the problem that the Sonoff POW sets its total kWh value back to 0 as soon as it exceeds 43 kWh. The result is that I no longer have the correct power consumption of the connected devices after the 43 kWh were exceeded. 

If you need more information just tell me which one.

arendst, 2018-07-12:
**Describe the bug**

_A clear and concise description of what the bug is._

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues and in the wiki
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Loreman01, 2018-07-12:
Describe the bug

I have the problem that the Sonoff POW sets its total kWh value back to 0 as soon as it exceeds 43 kWh. The result is that I no longer have the correct power consumption of the connected devices after the 43 kWh were exceeded.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!
X  Searched the problem in issues and in the wiki
X Hardware used : Sonoff POW
X Provide the output of command status 0 :

stat/282/STATUS {"Status":{"Module":6,"FriendlyName":"Sonoff","Topic":"282","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/282/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"xxx","OtaUrl":"https://xxx/sonoff.ino.bin","RestartReason":"Exception","Uptime":"36T03:03:23","StartupUTC":"","Sleep":0,"BootCount":14,"SaveCount":31,"SaveAddress":"F5000"}}
stat/282/STATUS2 {"StatusFWR":{"Version":"5.12.0e","BuildDateTime":"2018.03.19 19:48:55","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
stat/282/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId1":"POW02","SSId2":"","TelePeriod":300,"SetOption":"00000009"}}
stat/282/STATUS4 {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
stat/282/STATUS5 {"StatusNET":{"Hostname":"282-6436","IPAddress":"xxx","Gateway":"1xxx","Subnetmask":"255.255.0.0","DNSServer":"xxx","Mac":"68:C6:3A:95:79:24","Webserver":2,"WifiConfig":5}}
stat/282/STATUS6 {"StatusMQT":{"MqttHost":"xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_957924","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/282/STATUS7 {"StatusTIM":{"UTC":"Fri Feb 06 03:31:08 1970","Local":"Fri Feb 06 03:31:08 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1}}
stat/282/STATUS9 {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
stat/282/STATUS10 {"StatusSNS":{"Time":"1970-02-06T03:31:08","ENERGY":{"Total":0.739,"Yesterday":0.000,"Today":0.739,"Power":718,"Factor":0.94,"Voltage":231,"Current":3.292}}}
stat/282/STATUS11 {"StatusSTS":{"Time":"1970-02-06T03:31:08","Uptime":"36T03:03:23","Vcc":3.522,"POWER":"ON","Wifi":{"AP":1,"SSId":"POW02","RSSI":62,"APMac":"00:C0:CA:67:A6:02"}}}

To Reproduce
Take a Sonoff POW with total kWh close to 43 and connect a consumer so that the 43 kWh is exceeded. Meanwhile, look at the total values.

Expected behavior
Der total kWh wert sollte nicht zurück gesetzt werden sondern weiter zählen.

Screenshots
/

Additional context
/

arendst, 2018-07-12:
See release changelogs. Fixed in version 5.13.0

closed 2018-07-12 by Loreman01