arendst/Tasmota #3194: Sonoff NULL status and guess the real status of rele

Grhosso, 2018-07-12:
Hi there,

Everytime I reboot the system, the Sonoff send back the status NULL, but some switches are ON, and others are OFF.

Is there any way to guess the "real" status of the relé? In case is ON, change the status from NULL to ON, in case is OFF, changes from NULL to OFF??

I have tried some commands via http and MQTT, and always send NULL state after reboot system, so it is not able to guess the real status of the relay which had before.

Thank you for your support.

arendst, 2018-07-12:
**Describe the bug**

_A clear and concise description of what the bug is._

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues and in the wiki
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Grhosso, 2018-07-12:
I have been searching and I do not find this topic.

- HW: Sonoff Basic
- SW: Openhabian
- Status:
`{"POWER":"NULL"}`

Log in OH2:

`2018-07-12 11:03:12.897 [INFO ] [.eclipse.smarthome.model.script.RULE] - Estadoout (Type=StringItem, State=NULL, Label=null, Category=null)`

When Sonoff Basic is ON, and the Openhab restart, the STATE of the switch is NULL, and when I check by MQTT the state, it returns NULL. Should return ON, isn't it?

With command via HTTP, return ON. but through MQTT return SATATE= NULL. I know that state and power is not the same... but state should align with the status of the relé which is ON. In my opinion.

`http://IP_sonoff/cm?cmnd=status/power`

arendst, 2018-07-12:
Seems to me an openHab issue.

Grhosso, 2018-07-12:
I do not know, I guess, what I am trying to do is get a String from HTTP in order to get the POWER and not the STATE.

POWER is ON or OFF.
STATE is ON OFF or NULL.

But it seems that from OH I only get STATE through MQTT.

Jason2866, 2018-07-15:
After restart OH every state is NULL because OH dont store any states.
So if you want the states of the sonoffs you have to do a query for every device
With this you get the actual and correct states. Take a look in the wiki from Theo here
and for help for OpenHab ask here  https://community.openhab.org

ascillato2, 2018-07-21:
Hi

Seems that your question has been answered. Please, reopen this issue if needed. Thanks

closed 2018-07-21 by ascillato2