arendst/Tasmota #3195: Sensor DW1 notify only opening event

Hat3, 2018-07-12:
Hello to all

I use the RF Bridge Sonoff with last firmware with 2 door window sensors (DW1).

In the MQTT logs I receive only the message of the opening of the door and not of the closure.

What doing wrong?

arendst, 2018-07-12:
Nothing. The DW1 does not send a signal when closed as can be seen from the light of the DW1 unit.

Hat3, 2018-07-12:
The light after opening the door, flashes for a few seconds and then goes out.
Any solutions to know if the door is closed?

Jason2866, 2018-07-12:
Use a different sensor ;-)
The KERUI D026 sends open, close, tamper and low battery.

Hat3, 2018-07-12:
The KERUI D026 work with RF Bridge Sonoff (Tasmota)?

Jason2866, 2018-07-12:
YES 
16:19:34 MQT: tele/sonoff-8E8329/RESULT = {"RfReceived":{"Sync":9570,"Low":290,"High":940,"Data":"A53EEE","RfKey":"None"}}
16:19:35 MQT: tele/sonoff-8E8329/RESULT = {"RfReceived":{"Sync":9530,"Low":330,"High":950,"Data":"A53EE7","RfKey":"None"}}
16:19:37 MQT: tele/sonoff-8E8329/RESULT = {"RfReceived":{"Sync":9550,"Low":300,"High":940,"Data":"A53EEE","RfKey":"None"}}
16:19:40 MQT: tele/sonoff-8E8329/RESULT = {"RfReceived":{"Sync":9570,"Low":300,"High":930,"Data":"A53EE7","RfKey":"None"}}
16:19:45 MQT: tele/sonoff-8E8329/RESULT = {"RfReceived":{"Sync":9540,"Low":290,"High":940,"Data":"A53EEB","RfKey":"None"}}

Hat3, 2018-07-12:
Ok thank you!

closed 2018-07-12 by Hat3

salvq, 2018-10-09:
Hello Jason2866, I do have D026 Kerui and works great just never see low battery msg as my battery is OK. How do I setup, identify or simulate low battery status ? Is it still same mqtt topic as open / close ? Thanks

Jason2866, 2018-10-09:
Yes, it is still the same topic you get a different Values for Data.
You can try with a old battery or a power supply you can change the voltage...

salvq, 2018-10-09:
Thank you, will try

salvq, 2018-10-09:
Here is the value I am getting from D026 Kerui sensor for reference:
4E35E7 - closed
4E35EE - open
4E35EB - tamper
4E35E7 - low battery

GrexAut, 2019-06-25:
@salvq why you get other values than me from Kerui D026`?

ACFB2B => tamper
ACFB2E => open
ACFB2B => closed

tamper and closed has same values.

stefanlsd, 2020-07-16:
I wrote a post how I managed to at least turn the device into a closed sensor. https://snlr.us/s/sonoff-dw1-disappoints-harder-than-me/