arendst/Tasmota #3196: D6 GPIO12 : 70 SR04 Ech  / D7 GPIO13 : 69 SR04 Tri

pasqui7, 2018-07-12:
Hi, 
I want to realize this project:

https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-HC-SR04-Ultrasonic-Sensor

When I go to configuration of wemos, there are no :

70 SR04 Ech and 69 SR04 Tri 

The list arrive only at 

68 SerBr RX

Can you help me?

Thank you

Pako

Jason2866, 2018-07-12:
Take a look in user_config.h -> is it enabled?

pasqui7, 2018-07-12:
Hi janson. 
Firmware Updated ! 
Just boy this !

pasqui7, 2018-07-12:
Yes I can select actually 73 and 74 with version 6.1.0. 

In my opinion think that work fine without resistor becouse the device sr04 work with 5v. I can’t understand why we need to downgrade the voltage to 3.3v. 

I have read datasheet here :

https://cdn.sparkfun.com/datasheets/Sensors/Proximity/HCSR04.pdf

Jason2866, 2018-07-12:
You have to downgrade the voltage for the ESP chip input pin (GPIO) because the are build for 3.3 Volt!
5 Volt can and will probably damage your Wemos D1

Frogmore42, 2018-07-12:
There is some debate about whether the esp8266 can survive 5V on its GPIOs.  It is a 3V3 part, so it is safest to use the voltage divider to protect the esp8266 (not the SR04).

pasqui7, 2018-07-12:
Oh! I understand now ! Thank you for explaining! 

Now can you suggest to me which voltage divider I can use ?

I have one 1200ohm 2watt. = red red red gold

And one 2200ohm 2 watt = brown red red gold 

In your opinion, can I use this voltage divider ?

Thank you

Jason2866, 2018-07-12:
@Frogmore42 
I read some of this articels. The mess up circuit protections with device levels and there are "experts"
with statements "I have a device that works since ....." 
A good example are overclocked processors for working beyond the specs.
There lifetime is reduced and you will find articels and photos for this from real experts.
For such a cheap chip nobody does this expensive labor things...

pasqui7, 2018-07-12:
Ok now I search 0.125 watt. 

For now, I can try with this ?
 2 watt? Or doesn't works?

Frogmore42, 2018-07-12:
I agree that it is safest to NOT expect an esp8266 to survive 5V, especially on VCC but also on the GPIOs.

Those resistors should work fine, but the gold band indicates the tolerance (5%) not the power capability.  I would guess they are somewhere around 1/2w to 1/4w, which are the normal sizes.

Jason2866, 2018-07-12:
@pasqui7  Do you have it already at home? You can use it. It is okay. If you are going to buy some
look for 1/4 or 1/8 Watt Types

pasqui7, 2018-07-12:
Ok guys ! 

I can order this one :

Resistenze carbonio 1.2 K Ohm 1/4 Watt +/-5%, 1 https://www.amazon.it/dp/B01JGQ1UX4/ref=cm_sw_r_cp_api_NE3rBb2ZPG5TH

And this one :

Resistenza Resistore Resistore 2R2 2,2ohm 1/4W 5% carbone lotto di 10 pezzi https://www.amazon.it/dp/B076Q81DPY/ref=cm_sw_r_cp_api_GG3rBb7XRQ2HE

It’s fine for you ?

Jason2866, 2018-07-12:
You need 1k2 and 2k2.
First is ok. 
Second isnt. It is only 2.2 Ohm. You need 2200 Ohm (2k2 = 2200)

pasqui7, 2018-07-12:
Ok! Thanks!

Frogmore42, 2018-07-12:
Get this 
https://www.amazon.it/MakerHawk-Assortment-Resistenza-precisione-perfettamente/dp/B07D8RRW6T/ref=lp_10418684031_1_19?s=industrial&ie=UTF8&qid=1531411312&sr=1-19
Then you will lots of values for other things too.

pasqui7, 2018-07-12:
I had found it before, but if you see better, this kit have 2.2k but no 1.2k.

It's possible to use 1.0k and 2.2k?

pasqui7, 2018-07-12:
look this 👍 

1k2
https://www.ebay.it/itm/332570612190  

2k2
https://www.ebay.it/itm/232681930418

Jason2866, 2018-07-12:
What frogmore42 found will work. It doesnt matter...
Your ebay articles are the right ones.
Go and buy what you prefer more....

pasqui7, 2018-07-12:
ehi guys ! I have a question! 

but the 2 voltage divider, don't are so much expansive ? I mean: for reduce to 3.3V from 5V it's can do it with a resistor with less OHM! make sanse? Maybe you can use one voltage divider with 300 ohm! nope?

Frogmore42, 2018-07-13:
You can use this instead,
https://electronics.stackexchange.com/questions/329810/how-can-a-switching-diode-act-as-a-level-shifter

I have done this using just a diode 1N4148 worked okay and the internal pull up, but not on an esp8266.

Resistors are inexpensive compared to a real level shifter, so it depends on your perspective.

Designing a resistor divider is a classic (very simple) electrical engineering problem.  If you haven't learned how to do that in the first week of EE101, you would be behind.  If you are not in an electrical engineering class, there are many online tutorials that explain how they work.  There is math involved in figuring it out.  The total resistance needs to be within the driving capacity of the circuit doing the driving.  The bottom resistance needs to be low enough that the input impedance of the driven circuit does not change the ratio much.  The ratio needs to match the ratio of voltages you need.

pasqui7, 2018-07-16:
Hi guys! 
I have realized the project. But the distance it's not detected. 
it's blocked at 283cm.

Can you help me to trouble?

Frogmore42, 2018-07-16:
That is the maximum distance.  That indicates you have a greater distance or the connection is not correct, since the esp8266 is not seeing a response.

pasqui7, 2018-07-16:
Hey Frog! I'm used D8 and D0 pin! Now works fine! thank you!

ascillato, 2018-07-16:
Great!

If your issue is solved, please close it. Thanks :+1:

closed 2018-07-17 by ascillato2