arendst/Tasmota #3197: Local Staircase Functionality

Sandominion, 2018-07-12:
I am new user of Sonoff products. At least I managed to update the Sonoff TH with TASMOTA 5.13.x and configered the temp- and humility Sensor. My aim is to use it for watering my garden by giving power on the garden pump. Indipendend on the way of activation (remote or manualy with button on Sonoff) it shound switch off itself after 30min (even if the WiFi connection has got lost). Unfortunately I couldn't find any possibility to configure this behavier on the Sonoff itself. The discussed solutions I found all base on a procedure inside a central control (e.g. FHEM) which I don't have and don't want. Is there any chance to get without creating it on my one? Is this feature interesting for this community at all? Is there any other system to solve my issue easier (e.g. EspEasy)?
Kind regards
Sandominion

arendst, 2018-07-12:
Search wiki for timers and/or rules

Sandominion, 2018-07-12:
Thanks for the hint about Rules. Now it works - even without change of code ot config file.
I couldn't use the given example "10. Simple Staircase Timer" because it won't work with remote activation. My solution was to send via console the following commands (just in case somebody want's to solve a similar issue):
rule 1
rule 4
poweronstate 0
rule on Power1#State=1 do ruletimer1 1800 endon on rules#timer=1 do power1 off endon

closed 2018-07-12 by Sandominion