arendst/Tasmota #3198: Rule MQTT#Disconnected and one-shot detection

KPK75, 2018-07-12:
One-shot detection is not working on rule MQTT#Disconnected. Rule is executed every 10s when MQTT serwer is down and not once.

`Jul 12 12:45:14 sonoff11-0720 ESP-RUL: MQTT#DISCONNECTED performs "ruletimer1 60"`
`Jul 12 12:45:14 sonoff11-0720 ESP-RSL: stat/sonoff11/RESULT = {"RuleTimer1":60}`
`Jul 12 12:45:25 sonoff11-0720 ESP-MQT: Attempting connection...`
`Jul 12 12:45:25 sonoff11-0720 ESP-MQT: Connect failed to 192.168.5.201:1883, rc -2. Retry in 10 sec`
`Jul 12 12:45:25 sonoff11-0720 ESP-RUL: MQTT#DISCONNECTED performs "ruletimer1 60"`

ascillato, 2018-07-14:
Hi,

I could not reproduce this issue with last firmware **v6.1.1a**.

I test the rule you say (`on MQTT#DISCONNECTED do ruletimer1 60 endon`):

This is the complete console output:
- Start device
- Mqtt connection
- (I turn off my mqtt broker)
- Mqtt reconnection attempt
- Rule execution (once)
- Mqtt reconnection attempt
- Mqtt reconnection attempt
- Mqtt reconnection attempt
- (I turn on my mqtt broker)
- Mqtt reconnection OK
- Status 0 output added for information
- Rule information

```
00:00:00 Proyecto sonoff Cocina (Topic cocina, Fallback DVES_F7A6DD, GroupTopic sonoffs) Versión 6.1.1a-2_3_0
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como cocina-1757...
00:00:03 WIF: Conectado
00:00:03 HTP: Servidor web activo en cocina-1757 con dirección IP 192.168.1.32
00:00:05 MQT: Intentando conectar...
16:17:42 MQT: Conectado
16:17:42 MQT: tele/cocina/LWT = Online (Grabado)
16:17:42 MQT: cmnd/cocina/POWER = 
16:17:42 MQT: tele/cocina/INFO1 = {"Module":"Generic","Version":"6.1.1a","FallbackTopic":"DVES_F7A6DD","GroupTopic":"sonoffs"}
16:17:42 MQT: tele/cocina/INFO2 = {"WebServerMode":"Admin","Hostname":"cocina-1757","IPAddress":"192.168.1.32"}
16:17:42 MQT: tele/cocina/INFO3 = {"RestartReason":"External System"}
16:17:43 MQT: stat/cocina/RESULT = {"POWER1":"OFF"}
16:17:43 MQT: stat/cocina/POWER1 = OFF
16:17:51 MQT: tele/cocina/STATE = {"Time":"2018-07-14T16:17:51","Uptime":"0T00:00:17","Vcc":2.782,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":62,"APMac":"18:D6:C7:80:38:2C"}}
16:18:18 MQT: Falló Conexión a 192.168.1.2:1883, rc 8. Reintentando 10 seg
16:18:18 RUL: MQTT#DISCONNECTED performs "ruletimer1 60"
16:18:18 RSL: stat/cocina/RESULT = {"RuleTimer1":60}
16:18:29 MQT: Intentando conectar...
16:18:41 MQT: Intentando conectar...
16:18:53 MQT: Intentando conectar...
16:19:05 MQT: Intentando conectar...
16:19:17 MQT: Intentando conectar...
16:19:29 MQT: Intentando conectar...
16:19:41 MQT: Intentando conectar...
16:19:52 MQT: Intentando conectar...
16:20:03 MQT: Intentando conectar...
16:20:14 MQT: Intentando conectar...
16:20:25 MQT: Intentando conectar...
16:20:36 MQT: Intentando conectar...
16:20:41 MQT: Conectado
16:20:41 MQT: tele/cocina/LWT = Online (Grabado)
16:20:41 MQT: cmnd/cocina/POWER = 
16:21:04 KNX: Recibido desde 2.2.1 Comando Leer: 0 a Salida 1
16:21:22 MQT: stat/cocina/RESULT = {"POWER1":"OFF"}
16:21:22 MQT: stat/cocina/POWER1 = OFF
16:21:22 MQT: stat/cocina/RESULT = {"Command":"Unknown"}
16:21:22 MQT: stat/cocina/RESULT = {"Command":"Unknown"}
16:22:58 CMD: status 0
16:22:58 MQT: stat/cocina/STATUS = {"Status":{"Module":18,"FriendlyName":["Cocina"],"Topic":"cocina","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:22:58 MQT: stat/cocina/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:05:24","StartupUTC":"2018-07-14T19:17:34","Sleep":0,"BootCount":16,"SaveCount":81,"SaveAddress":"3F4000"}}
16:22:58 MQT: stat/cocina/STATUS2 = {"StatusFWR":{"Version":"6.1.1a","BuildDateTime":"2018-07-14T15:10:31","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:22:58 MQT: stat/cocina/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NetWireless","NetWireless"],"TelePeriod":300,"SetOption":["8E008009","55818000"]}}
16:22:58 MQT: stat/cocina/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":2596,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00002C0A","1F80A984","0C000010","23021796","00000000"]}}
16:22:58 MQT: stat/cocina/STATUS5 = {"StatusNET":{"Hostname":"cocina-1757","IPAddress":"192.168.1.32","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:F7:A6:DD","Webserver":2,"WifiConfig":5}}
16:22:58 MQT: stat/cocina/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F7A6DD","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:22:58 MQT: stat/cocina/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 14 19:22:58 2018","Local":"Sat Jul 14 16:22:58 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"08:47","Sunset":"18:28"}}
16:22:58 MQT: stat/cocina/STATUS10 = {"StatusSNS":{"Time":"2018-07-14T16:22:58"}}
16:22:58 MQT: stat/cocina/STATUS11 = {"StatusSTS":{"Time":"2018-07-14T16:22:58","Uptime":"0T00:05:24","Vcc":2.781,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":62,"APMac":"18:D6:C7:80:38:2C"}}}
16:23:02 CMD: rule
16:23:02 MQT: stat/cocina/RESULT = {"Rule1":"ON","Once":"ON","Free":468,"Rules":"on MQTT#DISCONNECTED do ruletimer1 60 endon"}
```

Sorry.

Seems that your issue is something else. May be that your broker is up again and down and up ?

KPK75, 2018-07-14:
What mqtt lib are you using?

ascillato, 2018-07-14:
TasmotaMQTT

KPK75, 2018-07-14:
Mine is pubsubclient and i think that's the problem. Time for a change. Thanks for the help.

ascillato, 2018-07-14:
Please, test it and let us know if that works for you.

KPK75, 2018-07-16:
Great job! Working as expected with pubsubclient.

closed 2018-07-16 by KPK75