arendst/Tasmota #3200: Gpios buttons internal pullups

vMeph, 2018-07-12:
Hi i have a esp8826 with tasmota firmware i have the generic configuration, by default i have a toggle switch that work correct via the web console, if i wanted to add a external switch to control also, my question is in what pin would i need to put the switch?
Another question is will i need a external pullup resistor, or do we have any pins with internal pullup already activated?

digiblur, 2018-07-12:
There are internal pull-ups on the GPIO pins and I know I've seen previous threads where even Theo showed the code of the pin being assigned a pull-up (it was a GPIO14 Sonoff basic thread).  With that said, I've had issues with the internal pull-ups being too weak or not working in some cases.  I save myself lots of headache and time but just doing my own pull-up resistor.

Frogmore42, 2018-07-12:
Part of it depends on your environment and how long wires you have.  There is both an art and a science to selecting an optimal pull-up.  The good news is that most of the time it doesn't matter that much.  Many MCUs implement internal pull-UPs.  Unfortunately, these are rarely well defined, typical might be 10K-50K ohms.

The pull-up needs to be small enough to overcome the leakage/impedance of the input and associated wiring.  The longer the wires, the more capacitance and the slower the signal rise time.

The pull-up needs to be large enough to allow something driving the line to be to actually drive the line low enough quickly enough.  The smaller the resistance of the pull-up the more current it takes to drive the signal low.

For something controlled by a physical switch, I would use something in the 1-2K ohm range.  That gives good noise immunity and power dissipation.

digiblur, 2018-07-12:
Was thinking about switching out the pull-up to something lower than the 10k I have now on a light that is located on a fan.  The fan switching on/off just makes too much noise for some reason and kicks the Sonoff, it was bad before the 10k and is only very rare now.  With your post I think I will swap things out.

vMeph, 2018-07-12:
yes that what i tought a external pullup will be the way to go

thanks for all input guys :)

closed 2018-07-12 by vMeph

deviant-studio, 2018-11-09:
i had lot of troubles with 10k. now i use 4.7k for all my sensors (temperature, buttons) and all seems good