arendst/Tasmota #3201: Wemos D1 mini Flash Tasmota

pasqui7, 2018-07-12:
Hi guys, can you tell me how can I put the device wemos d1 mini :

https://www.amazon.it/gp/product/B075HC84N8/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

into flash mode?

I need ft232rl device to flash it, or I can only install software driver and connect directly on my computer the device with micro usb cable?

thank you

**(Please, remember to close the issue when the problem has been addressed)**

digiblur, 2018-07-12:
You don't have to push any buttons on the Wemos d1 mini, just flash it.

pasqui7, 2018-07-12:
Great! Than only connect wemos with usb cable to computer.  Correct ?

digiblur, 2018-07-12:
Yep, plug it in, make sure you have the driver for the usb interface loaded, set your COM port and flash away.

pasqui7, 2018-07-12:
Great ! In Monday come back to you for updates! 
Thank you!

ascillato2, 2018-07-17:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-17 by ascillato2