arendst/Tasmota #3202: Sonoff ifan2 not connecting to wifi after update fail

donoo, 2018-07-12:
I got the latest ifan2 module yesterday and I tried to update it with Tasmota  using [SonOTA](https://github.com/mirko/SonOTA/blob/master/README.md). I used all default config and it was updated to 5.x.x version. It connected to wifi fine and everything was fine. Module was set to Sonoff Basic.

After that I used the update option from menu. I tried to update to latest version 6.1.0 to add ifan2 support. The update failed and after the device was restarted it is not connecting to wifi nor I m able to reset it using 40 sec reset option. Also it is not broadcasting it own wifi access point. 

Remote is also not working. When I click the wifi button on remote one single beep is played. Not sure how to reset it now?

Anyone faced similar issue to update to 6.1.0 or know how to reset without soldering it?

Thanks in advance!

maciboy, 2018-07-12:
Oh dear...
Pls join the club at #3177 

@arendst Honestly, I would suggest to pull this update, it generates so much pain :(

donoo, 2018-07-13:
Any suggestion for resetting it?

ascillato2, 2018-07-14:
Hi,

Duplicated issue. See #3177. Thanks

closed 2018-07-14 by ascillato2