arendst/Tasmota #3203: [Question] Reading Dutch Energy Meter (Kaifta) using P1 connector

RaymondMouthaan, 2018-07-12:
Hi Guys,

I have an Dutch energy meter [Kaifta MA105C], which measures electricity and gas. This meter has a P1 (rj11 connecter) to read serial data from it. I bought a serial -> usb cable and connected it to a windows laptop with putty installed. After setup the seriall connections with the following settings below I got the raw data from the meter ever 10 seconds.

**Settings:**
Port: Com3
Speed (baud): 115200
Data bits: 8
Stop bits: 1
Parity: None

Since Tasmota has support for serial, I was wondering whether it is possible to use a Wemos with Tasmota flashed to read the raw data and have it publish to a mqtt topic? So another words: Kaifta -> RJ11-USB-Cable -> Wemos (Tasmota) -> MQTT.

Jason2866, 2018-07-12:
You cant connect the serial - usb adapter to the usb port of the wemos.
I think you need a logic level serial to serial adapter., because i am pretty sure there arent 3.3 Volt levels.
If this is done it is easy connect tx to rx and activate serial bridge. You will get allt the messages via mqtt.
I use this too in a different use case. I have on a 3.3 V arduino the Rc-Switch sketch (decodes rf 433 signals) and receive it just this way.

RaymondMouthaan, 2018-07-12:
Jason, thanks for your response!

Could you please supply a link of this logic level serial to serial adapter? 
As I understood, my current rj11 to usb cable with build in FT232R chip (inverted rxd signal) will not work with Wemos. So your solution would be to make a DIY cable with a RJ11 connector one side, then the logic level serial to serial adapter and possibly a circuit to invert RxD. And then connect to one of the gpio's of the wemos?

Jason2866, 2018-07-12:
Shifter
https://www.aliexpress.com/item/IIC-I2C-level-conversion-module-5V-to-3V-System-Compatible-Module-Dedicated/32443447640.html?spm=2114.search0104.3.15.195b2455sMOSfO&ws_ab_test=searchweb0_0,searchweb201602_3_10320_10152_10151_10065_10321_10344_10068_10342_10547_10343_10322_10340_10548_10341_10696_10084_10083_10618_10304_10307_10820_10821_10302_10846_10843_10059_100031_10319_10103_524_10624_10623_10622_10621_10620-10152,searchweb201603_1,ppcSwitch_5&algo_expid=88a004b1-2c3d-47bc-a29c-03962f112f51-2&algo_pvid=88a004b1-2c3d-47bc-a29c-03962f112f51&priceBeautifyAB=0

Jason2866, 2018-07-12:
And of the early beginning of the electronics to invert a ttl signal you can use a 74HCT04 😊

RaymondMouthaan, 2018-07-12:
yes or even simpler a transistor like this http://i.imgur.com/5waRPTp.jpg or a  MAX232, 7407 IC :-)

Jason2866, 2018-07-13:
Go  and try it with the transistor :-)
MAX232 wont work here because this chips convert levels from RS232 serial to TTL
This levels are not TTL nor 3.3V. As far as i remember the are from -3V to -25V "0" and +3V to +25V "1"

Jason2866, 2018-07-13:
This is what i geht over the serial port in my use case(Arduino nano output TX to Wemos Input RX)
07:51:42 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377620"}
07:51:42 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377620"}
07:51:42 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377620"}
07:51:42 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377620"}
07:51:42 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377620"}
07:51:42 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377620"}
07:51:42 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377620"}
07:51:43 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377620"}
07:51:43 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1377631"}
07:51:52 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1380692"}
07:51:52 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1380692"}
07:51:55 MQT: tele/sonoff-17DBAE/RESULT = {"SerialReceived":"1381460"}

Jason2866, 2018-07-13:
On the hardware side, mybe easiest and the safest solution
optocoupler like the 4n35

Jason2866, 2018-07-13:
RTS Pin (P1) is TTL Level, so it is 100% correct and safe to connect it with 5V from the Wemos.
5V on the Wemos delivers easily the current that you need for that...
This little circuit will do the job, as long the wires are not too long.

RaymondMouthaan, 2018-07-13:
Thats the confirmation I was looking for 👍  ... I'll let you know if finally works :-)

Jason2866, 2018-07-13:
Your welcome.
I have only a dumb old Energy Meter :-(

RaymondMouthaan, 2018-07-13:
Not possible to request a new meter? Here in the Netherlands they replace all dumb meters by Smart meters in phases. Fortunately my (new) house (i just moved) has a smart one :-)

Jason2866, 2018-07-13:
Yes, the do BUT here in my region of germany the charge me fot that (one time and monthly)

Jason2866, 2018-07-13:
Just moved and time for playing around with "little" projects?  :-)

RaymondMouthaan, 2018-07-13:
Actually I should been busy with other things, but Home Automation is just to fun and to addictive, hahaha

RaymondMouthaan, 2018-07-13:
@Jason2866 

I did the hardware part with the two resistors, transistor and Wemos and setup the Wemos like this:

[image]

Below the results in Tasmota Console:
```
14:46:34 CMD: /KFM5KAIFA-METER
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-3:0.2.8(42)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-0:1.0.0(180713144734S)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-0:96.1.1(4530303235303030303639363432393136)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:1.8.1(001433.161*kWh)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:1.8.2(001779.004*kWh)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:2.8.1(000000.000*kWh)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:2.8.2(000000.000*kWh)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-0:96.14.0(0002)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:1.7.0(00.423*kW)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:2.7.0(00.000*kW)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-0:96.7.21(00000)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-0:96.7.9(00000)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:99.97.0(1)(0-0:96.7.19)(000101000001W)(2147483647*s)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:32.32.0(00000)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:32.36.0(00000)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-0:96.13.1()
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-0:96.13.0()
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:31.7.0(002*A)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:21.7.0(00.418*kW)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 1-0:22.7.0(00.000*kW)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-1:24.1.0(003)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:34 CMD: 0-1:96.1.0(4730303332353631323736373836373136)
14:46:34 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:35 CMD: 0-1:24.2.1(180713140000S)(02443.898*m3)
14:46:35 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
14:46:35 CMD: !8F54
14:46:35 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
```

So basically Tasmota receive the data perfectly from the energy meter, but every line is interpreted as a CMD, which results in Command Unknown. 

I've tried changing the SerialDelimiter to \n and \r, but this didn't change anything. I guess I am missing something here ...?

btw. for the moment my mqtt broker is still down.

Jason2866, 2018-07-13:
Perfect job, sometimes electronic works as wanted.

RaymondMouthaan, 2018-07-13:
It works like you said 👍🏽👍🏽👍🏽, I'll play around a little more with the SerialDelimiter to see what works best. Then parse the data with node-red, but for that I need to set my docker swarm back online 😎.

Thanks again for all your help! 👍🏽

A soon as I got time I will update Tasmota Wemos Wiki accordingly 😇.

closed 2018-07-13 by RaymondMouthaan

Jason2866, 2018-09-30:
Hi @svh1985 
you can edit the wiki. Just do it!

svh1985, 2018-10-18:
@Jason2866 haha of course I can, but I actually want to use his proposed wiki page to build it because I can't get it to work.

neographikal, 2018-11-16:
> Hi Raymond,
> 
> Just wondering if you had the chance to update the Wiki?
> 
> I used this firmware without the need of any transistors or resistors in the past:
> https://github.com/jantenhove/P1-Meter-ESP8266
> 
> I would love to get it working on Tasmota.

I wrote that port, but it's not perfect, CRC check is missing because I could not get it to work. Would love support in Tasmota as well.

RaymondMouthaan, 2019-01-04:
Initial Tasmota wiki page can be found [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-P1-Energy-Meter)

niekvlessert, 2019-01-16:
Hello all, interesting read. I'm not that deep into electronics like this, but I would like to give this a shot. However I have a tiny 5 to 3.3v converter board from aliexpress. What if I put this inbetween? This would mean there's no inversion, would the connection be possible like that?

RaymondMouthaan, 2019-01-17:
@niekvlessert, as understood from various websites the invert depends on the energy meter. I would just try the tiny 5v to 3,3v converter and check whether you received readable serial data within the Tasmota web console. Otherwise you could easily buy the few parts (transistor and resistor) from a local electronics store.

niekvlessert, 2019-01-17:
@RaymondMouthaan, thanks. I'll also have a look what exact type I have, I'd say I could find online if it is inverted or not, I missed that there are types that don't do inverted. I was wondering, what if it is still inverted, it would be possible for a program  to interpret every incoming 0 as a 1 and the other way around...? I can't find some setting or anything about it, so I guess Tasmota doesn't support that.

RaymondMouthaan, 2019-01-17:
@ascillato2 @mike2nl do you know whether it supported by Tasmota to invert serial bit data?

gemu2015, 2019-08-21:
@RaymondMouthaan and others interested

i am looking for someone to check a driver with your type of p1 meter interface.
i developed a driver for SML/OBIS that works with nearly all german house energy meters.
it directly decodes all messages and sends them as MQTT and has a Web Interface for all values.
it is now in use for about 2 years and a lot of german tasmota users already use it.
the last version is much easier to configure than previous versions.

if you are interested here is a further description:

[sml_obis_tasmota.txt](https://github.com/arendst/Sonoff-Tasmota/files/3524411/sml_obis_tasmota.txt)

meingraham, 2019-08-24:
@gemu2015 

Is it too soon for you to add `>M` to the Scripting wiki along with the sample script? Any value in adding the other script you use to configure the different types of devices or is that just too "specialized?

Cheers!

Mike

gemu2015, 2019-08-24:
@meingraham 
its too soon.  i want to wait until the driver may be merged. 
the driver is extremely flexible and can handle all kind of meters. but it needs (in the easy form) scripter to define the meters. otherwise you have to define the meters with #defines in the driver.
in germany we have dozens of different meters and you have to adapt the descriptors in the driver itself if your meter is not already in the list.
in the netherlands we have a more uniform type of interface (p1) 
i have no knowledge about other countries
the driver simply receives serial streams and compares them for patterns. it then extracts the values from the stream.
through the descriptors you can easily define the patterns and values, scale and format them.
this is however very different from other sensor drivers in Tasmota.

 Cheers

Gerhard

meingraham, 2019-08-24:
Understood.

Once you have the driver ready, a "descriptors cookbook" could be prided via the wiki to add information like you have set up in this script example - https://github.com/arendst/Sonoff-Tasmota/issues/6245#issuecomment-522207609

Mike

hdoedens, 2019-10-23:
I have a similar setup to @RaymondMouthaan I also had the serial messages being interpreted as commands. so I issued the `serialsend` command. I would like the entire raw telegram to be send to MQTT, but the message is broken up into three parts like so:
```
{"SerialReceived":"/XMX5LGBBFFB231222256

1-3:0.2.8(42)
0-0:1.0.0(191023212358S)
0-0:96.1.1(4530303034303031363930363534343134)
1-0:1.8.1(007881.115*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:1.8.2(008517.536*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.227*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00015)
0-0:96.7.9(00005)
1-0:99.97.0(5)(0-0:96.7.19)(191001151227S)(0000002058*s)(191001092911S)(0000004972*s)(190929200954S)(0000000475*s)(190929183545S)(000000 ...

{"SerialReceived":"0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(001*A)
1-0:21.7.0(00.227*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303032333430313933343439313135)
0-1:24.2.1(191023210000S)(03247.118*m3)
"}

{"SerialReceived":"!E9E7
"}
```
My question is: Did you have this same issue and if so, how did you solve it? I have tried several `serialdelimiter`s but none of them work.
The reason to have an intact telegram is to [feed DSMR-Reader with raw information over MQTT](https://github.com/dennissiemensma/dsmr-reader/issues/728).

RaymondMouthaan, 2019-10-23:
@hdoedens - yes i do have it broken up into several messages .. these message are read in node-red where i joint the messages in to one .. from there i post them to the rest api of DSMR reader

hdoedens, 2019-10-23:
@RaymondMouthaan thanks for the comment. I also searched and tried a lot of settings. I guess there is more to it than a setting...

svh1985, 2020-02-01:
Is it possible to exclude the break line? My payloads look like this:
```
{"SerialReceived":"1-0:22.7.0(00.000*kW)
"}
```
Home Assistant doesn't see this as valid JSON.

ascillato, 2020-02-01:
Yes, check serialdelimiter command. Thanks.

svh1985, 2020-02-01:
The serialdelimiter is set to 10

ascillato, 2020-02-01:
Serialdelimiter 128

svh1985, 2020-02-05:
Still doesn't work 😞