arendst/Tasmota #3204: Rules - Inter-lock Sonoff 4CH Pro Channels (Help Wanted)

meingraham, 2018-07-12:
I have several ceiling fans which I want to connect to my home automation hub (openHAB) and be able to control via the mobile app or via Google Assistant.

A Sonoff 4CH Pro is ideal for the four functions each of them has: the fan light and three fan speeds.I cannot use the 4CH Pro inter-locking mode (only one relay turns on at a time based on the button input) because some fans require more than one contact to be connected simultaneously for achieving some of the speed settings. Also, If I want to turn on the fan and also have the fan light on, the 4CH inter-locking mode would not allow those outputs on simultaneously. For these reasons, I am using the 4CH switches in self-locking mode... even when the particular brand of fan uses a single contact scheme for the fan speed.

Our ceiling fans are manufactured by three different companies: Hunter, Hampton Bay, & Harbor Breeze. These fans use two types of 4-position speed control switches. The connections for different types of fans:

Hunter
Low: Ch.1
Med: Ch.1 & Ch. 2
Hi: Ch.3
Light: Ch. 4

Harbor Breeze
Low: Ch.1
Med: Ch. 2
Hi: Ch.3
Light: Ch. 4

I haven't cracked open my Hampton Bay wiring yet, but I'm sure the fan speed switch falls into one of the two categories - single-contact (e.g., Harbor Breeze) or multi-contact (e.g. Hunter).

I would like to set up rules in the TASMOTA firmware to "prohibit" certain combinations of relays from being turned on simultaneously. The rules would vary slightly for each brand of fan connected.
Hunter: inter-lock Ch.1, & Ch. 3
Harbor Breeze: inter-lock Ch.1, Ch.2, & Ch. 3

1. Can "inter-lock" rules be implemented in TASMOTA?
2. I've never written a TASMOTA rule... so would appreciate any and all examples.

Thanks in advance.

Mike

localhost61, 2018-07-15:
Harbor Breeze ceiling fan wiring diagram, maybe [there](https://www.techvi.com/14-more-wiring-diagram-for-a-harbor-breeze-ceiling-fan-graphics/wiring-diagram-for-a-harbor-breeze-ceiling-fan-for-harbor-breeze-ceiling-fan-switch-within-wiring-diagram-harbor-on-techvi-com-photos/)

meingraham, 2018-07-16:
I need a schematic of the internals of the fan (the internal speed capacitor circuits and how they are connected to the motor windings. The mains wiring is straightforward and most of these diagrams on the net are for that anyhow.

Regardless... I'm just want to add a failsafe via TASMOTA rules. But it doesn't seem anybody knows if it's possible to set up a 4CH in this manner nor how to do what I'm asking. Neither this post nor a similar one on the Sonoff Google Group have elicited any responses to my rules configuration question.

I am waiting to receive four new 4CH Pro switches. Once I get them I'll flash one of them and start trying this by brute force. Good luck to me :(

Mike

meingraham, 2018-07-19:
Success! Fortunately, I only need to inter-lock three channels... for which there are just enough TASMOTA rule "slots" available. Each rule uses 266 characters of the available 511.

Here are the three rules. Each rule is a variation of the toggling rule in the TASMOTA Rules wiki page examples. This rule set is for the Hunter fan which has the need to turn on two channels for Medium (Rule 2) but only one channel for either low or high. For my Harbor Breeze and Hampton Bay fans, each fan speed needs at most one relay latched. That rule set is a pretty straightforward variation of this set of rules.

I broke the rule syntax up into multiple lines for readability. The rule needs to be entered in the TASMOTA Console as a single line. I have included the "single line" rules as well for proper reference.

> Rule1 on BUTTON1#State do
>         Event toggling1=%var1%
>       endon
>       on event#toggling1<1 do
>         Backlog
>             Power1 ON
>           ; Power2 OFF
>           ; Power3 OFF
>           ; Event setvar1=1
>       endon
>       on event#toggling1>0 do
>         Backlog
>             Power1 OFF
>           ; Power2 OFF
>           ; Power3 OFF
>           ; Event setvar1=0
>       endon
>       on event#setvar1 do
>         var1 %value%
>       endon
> 
`Rule1 on BUTTON1#State do Event toggling1=%var1% endon on event#toggling1<1 do Backlog Power1 ON; Power2 OFF; Power3 OFF; Event setvar1=1 endon on event#toggling1>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; Event setvar1=0 endon on event#setvar1 do var1 %value% endon`

> Rule2 on BUTTON2#State do
>         Event toggling2=%var2%
>       endon
>       on event#toggling2<1 do
>         Backlog
>             Power1 ON
>           ; Power2 ON
>           ; Power3 OFF
>           ; Event setvar2=1
>       endon
>       on event#toggling2>0 do
>         Backlog
>             Power1 OFF
>           ; Power2 OFF
>           ; Power3 OFF
>           ; Event setvar2=0
>       endon
>       on event#setvar2 do
>         var2 %value%
>       endon
> 
`Rule2 on BUTTON2#State do Event toggling2=%var2% endon on event#toggling2<1 do Backlog Power1 ON; Power2 ON; Power3 OFF; Event setvar2=1 endon on event#toggling2>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; Event setvar2=0 endon on event#setvar2 do var2 %value% endon`

> Rule3 on BUTTON3#State do
>         Event toggling3=%var3%
>       endon
>       on event#toggling3<1 do
>         Backlog
>             Power1 OFF
>           ; Power2 OFF
>           ; Power3 ON
>           ; Event setvar3=1
>       endon
>       on event#toggling3>0 do
>         Backlog
>             Power1 OFF
>           ; Power2 OFF
>           ; Power3 OFF
>           ; Event setvar3=0
>       endon
>       on event#setvar3 do
>         var3 %value%
>       endon
> 
`Rule3 on BUTTON3#State do Event toggling3=%var3% endon on event#toggling3<1 do Backlog Power1 OFF; Power2 OFF; Power3 ON; Event setvar3=1 endon on event#toggling3>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; Event setvar3=0 endon on event#setvar3 do var3 %value% endon`

And don't forget to turn the rules on:
`Backlog Rule1 1; Rule2 1; Rule3 1`

Regards.

Mike

closed 2018-07-19 by meingraham

ascillato2, 2018-08-23:
@meingraham You can concatenate the rules. The 3 slots are there but each rule that start with ON is independent from each other. So you can add more triggers if you want.

meingraham, 2018-08-23:
@ascillato2 OK, makes sense. I'll make a note of this... but since my three rule combined exceed 511 characters, I probably keep these three separate for "easier" management. But, if I ever need a slot, I now know that I can combine two of them to free up a rule slot. Of course, it's a good tip for future need. Thanks.

ascillato, 2018-08-23:
You have 511 per set.

RULE1 has 511 + RULE2 has 511 + RULE3 has 511 = 1533 characters for rules :smile:

ascillato, 2018-08-23:
the sets allow you to make some rules with the RULE_ONCE activated (in one set) and other rules without that (other sets).

meingraham, 2018-08-23:
Oh, so it's not segmented to 511 separately? It's a combined 1533? The note that is returned when a rule is defined makes it seem like there's a 511 limit max on any one rule.

ascillato, 2018-08-23:
> it's not segmented to 511 separately?

no

> It's a combined 1533?

yes :smile: 

> The note that is returned when a rule is defined makes it seem like there's a 511 limit max on any one rule.

That is remaining of each set. RULE1, RULE2 and RULE3

remember, if you use the command RULE, Tasmota will use RULE1.

meingraham, 2018-08-23:
I tried this:

Enter a "short" rule. The response in the Console is:

`MQT: stat/sonoff-keurig_toggle/RESULT = {"Rule1":"OFF","Once":"OFF","Free":274,"Rules":"BUTTON2#State do Event toggling2=%var2% endon on event#toggling2<1 do Backlog Power3 OFF; Power1 ON; Power2 ON; var1 0; var2 1; var3 0 endon on event#toggling2>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; var1 0; var2 0; var3 0 endon"}`

In other words, the rule is "echoed" back and the "Free" value is reflected.

However, if I enter the "long" rule

`Rule on BUTTON1#State do Event toggling1=%var1% endon on event#toggling1<1 do Backlog Power2 OFF; Power3 OFF; Power1 ON; var1 1; var2 0; var3 0 endon on event#toggling1>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; var1 0; var2 0; var3 0 endon on BUTTON2#State do Event toggling2=%var2% endon on event#toggling2<1 do Backlog Power3 OFF; Power1 ON; Power2 ON; var1 0; var2 1; var3 0 endon on event#toggling2>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; var1 0; var2 0; var3 0 endon on BUTTON3#State do Event toggling3=%var3% endon on event#toggling3<1 do Backlog Power1 OFF; Power2 OFF; Power3 ON; var1 0; var2 0; var3 1 endon on event#toggling3>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; var1 0; var2 0; var3 0 endon`

the response is

`MQT: stat/sonoff-keurig_toggle/RESULT = {"Rule1":"OFF","Once":"OFF","Free":511,"Rules":""}`

as if thought it "threw away" my rule definition. Note that I specified "Rule" when I sent the long rule rather than "Rule1".

I checked all three rules - they are all "empty". This seems to imply that it doesn't like rules longer than 511.

ascillato, 2018-08-23:
Each set is 511

ascillato, 2018-08-23:
you set up to 511 for each set. all rules inside all set will be executed if the trigger is met

meingraham, 2018-08-23:
Where is that 511 default limit set? I didn't see that in the config header.

meingraham, 2018-08-23:
OK. Net-net... I can "fill up" each rule slot with as many triggers and actions as will fit in each rule slot. As long as a contiguous On/EndOn fits, just keep filling up the slot. Tip noted. Thanks.

creativtransfer, 2020-02-23:
How do I detect a long press on a sonoff 4ch? 

Currently, I am using this rule to toggle all four relais at once, but I haven't figured out how to accomplish this with a long press, so that a short press of button1 will toggle power1 only!

`Rule1 on BUTTON1#State do Event toggling1=%var1% endon on event#toggling1<1 do Backlog Power1 ON;Power2 ON; Power3 ON; Power4 ON; var1 1 endon on event#toggling1>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; Power4 OFF; var1 0 endon
`
BUTTON1#State=3 does not work. I already tried that. 

Is it possible at all to get a long press event on a sonoff 4ch? 
So that a short press will work as before but a long press will turn on/off all relais.