arendst/Tasmota #3205: Help with SDM630-MODBUS and Sonoff Basic

hahmann, 2018-07-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : SDM630 + Sonoff Basic
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
22:32:17 CMD: status 0
22:32:17 MQT: stat/irrigation_valve_2/STATUS = {"Status":{"Module":1,"FriendlyName":["irrigation_valve_2"],"Topic":"irrigation_valve_2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

I'm trying to connect SDM630-MODBUS to Sonoff Basic, I have the following configuration for gpio:
22:33:52 CMD: gpio
22:33:52 MQT: stat/irrigation_valve_2/RESULT = {"GPIO1":"78 (SDM630 Rx)","GPIO3":"77 (SDM630 Tx)","GPIO4":"0 (None)","GPIO14":"0 (None)"}

But could not find anything on the wiki showing the connection between SDM630 and sonoff serial interface, so not sure I'm missing any hardware to do the RS485 to serial on sonoff.

Really appreciate any help here.

Thanks,
Gabriel.

Jason2866, 2018-07-13:
You need on the hardware side something like this:
https://www.aliexpress.com/item/TTL-to-RS485-Power-Supply-Converter-Board-3-3V-5V-Hardware-Auto-Control-Module-For-Arduino/32852445547.html?spm=2114.search0104.3.8.25cf7ef6uZpUdg&ws_ab_test=searchweb0_0,searchweb201602_2_10320_10152_10151_10065_10321_10344_10068_10342_10547_10343_10322_10340_10548_10341_10696_10084_10083_10618_10304_10307_10820_10821_10302_10846_10843_10059_100031_10319_10103_524_10624_10623_10622_10621_10620,searchweb201603_2,ppcSwitch_5&algo_expid=37782e0c-481a-4aa5-9c66-cd0a21b484bf-1&algo_pvid=37782e0c-481a-4aa5-9c66-cd0a21b484bf&priceBeautifyAB=0

hahmann, 2018-07-16:
Perfect, thank you @Frogmore42 and @Jason2866, I managed to make the connection work using a MAX3486 I had here and did my own adapter with automatic TX/RX, it is working properly and as expected reporting to sonoff-tasmota.

closed 2018-07-16 by hahmann