arendst/Tasmota #3206: MagicHome: Add a normal wall switch

titomontes, 2018-07-13:
Hello everyone

I´ve been looking to add a dimmer to my led strips (changing hardware and all), but I´m not sure if I´ll be able to keep my normal wall switch as I have it with a sonoff (connected to GPIO14). 

Is there a way to use a GPIO on the MagicHome controller and configure it as a normal switch?

Jason2866, 2018-07-13:
There are several different named MagicHome Controllers. Which one?
And yes in general this is possible as similar with a sonoff.
Easiest go and get a MagicHome 801
https://github.com/arendst/Sonoff-Tasmota/wiki/H801

ascillato2, 2018-07-21:
Hi

Seems that your question has been answered. Please, reopen this issue if needed. Thanks

closed 2018-07-21 by ascillato2