arendst/Tasmota #3207: Uptime Reset after loosing Wifi Signal at sOnOffPow 1 + 2

LurchiStromberg, 2018-07-13:
HI!
I recognized, that the Uptime is resetted, after the Router restarts or lost Wifi Signal.

I think it would be better to be able to set a Starttime, when Pow starts Monitoring, that wont reset after loosing Wifi or else.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : 
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-07-13:
I think it should be possible with a rule and using a variable
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#5-use-of-variables-and-tele-

LurchiStromberg, 2018-07-13:
Thank you for your Answer!
I am not that much into that Stuff...

Can you give me a hint how that should work connected to my "Problem" with the Uptime?
Its funny that nobody else recognize that "Issue"
Because.
Isnt it an urgent Information how long the Pow is running already if i want to calculate the periode and usage of Power recorded?
Sure i can do that at FHEM log for example but it would be nice, if Tasmota would support that out of stock.
Just in the Overview Page to see, total kw/h and since when its recorded. 
Urgent and useful Information if you ask me.

Frogmore42, 2018-07-13:
Log showing the issue?
Status 0

You probably have it configured to reset when it loses connection.  Uptime is the amount of time the device has been up, which is working as designed.

I use the POW for measurements and have had no issue other than on a power failure it can sometimes go backwards, but that is a pretty rare occasion and saving to flash more frequently would wear it out faster.

The POW provides the power used today, and total.  If you want more detail you reply need to store the data in your home automation system, or something like emoncms, or influxdb.

LurchiStromberg, 2018-07-13:
Well sure, like I said, its possible in FHEM.

But you have an Overview in Tasmota and i think the Job is easy to do (of course not for me) to have a Point, what shows since what date / time the total ammount of kw/h is running.

arendst, 2018-07-14:
No need to store anything. Just change wificonfig to WIFI_WAIT (5) and the device will not restart when your router is lost and uptime will keep counting...

LurchiStromberg, 2018-07-14:
Oh, is it really that easy?
Looks like i have to learn a lot still :(

Frogmore42, 2018-07-14:
That will work till there is a power failure.  When there is a power failure, the POW will reset, so uptime will reset to 0. If it has access to the internet when power comes back it will know what time it really is, but it currently only logs that information.  If it doesn't have access to the internet, it has no way of knowing what time it really is.

Jason2866, 2018-07-14:
@Frogmore42 I tried it with rules. -> failed. Maybe it is a easy solution but i dont see it.
I am still thinking it should be possible (store it in a variable and read it)

ascillato2, 2018-08-27:
Hi,

Closing as the original topic of the issue (losing uptime because tasmota resets after wifi lost) can be solved as explained by wificonfig 5 command

Please, ask to reopen if needed. Thanks

closed 2018-08-27 by ascillato2