arendst/Tasmota #3208: commands not work

jsalinaszgz, 2018-07-13:
Yesterday compile 6.1 master tasmota with 2.4.1 and voltageset, currentset, powerset do not work.
Sonoff basic R2

Frogmore42, 2018-07-13:
Please fill in all of the information requested in the template or no one will be able to help you other than guessing that you are sending the commands incorrectly to a device that does not support them.  The basic does not have power measurement circuitry built in, so none of those commands make any sense for it.

jsalinaszgz, 2018-07-13:
i am a beginner and my english is very bad too. give me a few days to do all right.
i have a sonoff basic with a pzem 004 in gpio 1 and 3. i can to see voltage, current and power data in browser and console but not calibrate them.
i need a sonoff pow to conect a pzem and calibrate this?
thanks

Frogmore42, 2018-07-13:
I don't have a pzem 004 so I don't know how it works, but using it with a Sonoff basic should be fine.  I also don't know its capabilities, but I am sure there are others here who do know about them.

arendst, 2018-07-14:
As the pzem is a dedicated energy monitor device calibration in tasmota is currently not supported.

jsalinaszgz, 2018-07-14:
Thanks Frogmore42.
It is understood Theo thanks for creating tasmota and for all your work.

ascillato2, 2018-07-14:
Hi,

Seems that your question has been addressed. Closing this issue. Please reopen it if needed. Thanks.

closed 2018-07-14 by ascillato2

h4wk3y, 2019-02-11:
Hi, please, is in tasmota some option to divide readed current from PZEM-004t? When I use 2 round of wire throught measuring coil (for wider use of measuring range), then its necessary to divide current by number of rounds... Is it posible, please? Thank you

Frogmore42, 2019-02-12:
See Community Forum and Discord Chat for additional user experience
On the wiki, put it into desktop mode to see the link if using a mobile device.

Do not use a closed issue to ask a support question.