arendst/Tasmota #3211: Sonoff Device restarts in a loop

tango156157, 2018-07-13:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :

Hello Everyone,
I have installed Tasmota 6.1.0 firmware in my ESP 8266 and sonff basic and dual. When the device is connected to the Wifi, it works fine. But when the wifi connection is down or Wifi is switched off,,  the device starts looking for the wifi and the device restarts after every 3 mins. 

It restarts every 3 mins and my ights connected to it gets flickered every time the Sonoff restarts. I want to deactivate my Wifi manager so that my device won't restart. I want that my device continuously search for the Wifi. Is there any setting in the firmware where I can activate or deactivate the Wifi manager? Or when I install the firmware I can change the settings. If anybody has the solution to my problem, kindly help. Any assistance will be very useful. 

WIF: Connecting to AP1 QT_Lab in mode 11N as sonoff-3994...
13:52:45 WIF: Connect failed as AP cannot be reached
13:52:46 WIF: Connecting to AP2 QT_Lab in mode 11N as sonoff-3994...
13:52:53 WIF: Connect failed as AP cannot be reached
13:52:53 WIF: Wifi Manager active for 3 minutes
13:52:53 HTP: Web server stopped
13:52:53 HTP: Web server active on sonoff-3994 with IP address 192.168.4.1

Thanks

Frogmore42, 2018-07-13:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
Look at WiFi config I am sure one of the many options there will do what you want.

ascillato2, 2018-07-16:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2