arendst/Tasmota #3213: Help - just will not build latest version!

Mikeinnc, 2018-07-13:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : Sonoff basic
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE ? Sorry not sure?
```
Previously sucessfully flashed Tasmota 5.14.0 on devices. Have some new Sonoff Basic devices, so downloaded new version as the latest zip file, and unzipped to a directory. I'm using PlatformIO in Atom to build - same FDTI hardware etc as before This is the new version - the older version built perfectly. Now it fails to build every time with following error messages:
Compiling .pioenvs\sonoff\src\core_esp8266_timer.c.o
Compiling .pioenvs\sonoff\src\core_esp8266_wiring_digital.c.o
sonoff\core_esp8266_timer.c:21:28: fatal error: wiring_private.h: No such file or directory

#include "wiring_private.h"
^
sonoff\core_esp8266_wiring_digital.c:22:28: fatal error: wiring_private.h: No such file or directory

I have tried this numerous times - downloaded; unzipped and set the project - but every time is the same! Always the same error. Any help would be gratefully received!

Many thanks

Mike

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-07-13:
Tried no problem build successful. I used latest development 6.0.1a

reloxx13, 2018-07-13:
the pointed line is not the correct error line, the compiler cannot find the correct lines cuz of imports etc.

if you always copy your user_config back in the freshly downloaded dir, youll have an error in it.

Mikeinnc, 2018-07-14:
OK, I have tried these fixes - used the clean process; updated absolutely **everything;** removed the directories with the unzipped files in; downloaded and unzipped the latest version; updated the now virgin user_config.h file to reflect my system - and **still** get the same problem with reference to not being able to find wiring_private.h. I even went to another computer; loaded a new Atom instance and installed PlatformIO - but this failed as the PlatformIO would **NOT** render correctly inside Atom and so it is totally unusable!! You couldn't believe it if you tried!!!! (and, yes, I've reported that as a bug in PlatformIO area). As I've said, the previous version of Tasmota built and uploaded perfectly. I know **nothing** is wrong with my process _or_ the actual new board - I had been able to upload the older version on to it. It was only after I updated to the latest version that all these errors started. Thanks for the assistance so far - any clues as to what I might try next would be gratefully received!!

closed 2018-07-14 by Mikeinnc

Mikeinnc, 2018-07-14:
So, I tried uninstalling Atom and PlatformIO and reinstalling. Exactly the same problem. Here's the error message I receive:

Compiling .pioenvs\sonoff\src\core_esp8266_timer.c.o
Compiling .pioenvs\sonoff\src\core_esp8266_wiring_digital.c.o
sonoff\core_esp8266_wiring_digital.c:22:28: fatal error: wiring_private.h: No such file or directory
************************************************************************
* Looking for wiring_private.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:wiring_private.h"
* Web  > https://platformio.org/lib/search?query=header:wiring_private.h
*
************************************************************************ 
#include "wiring_private.h"
^
sonoff\core_esp8266_timer.c:21:28: fatal error: wiring_private.h: No such file or directory 
************************************************************************
* Looking for wiring_private.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:wiring_private.h"
* Web  > https://platformio.org/lib/search?query=header:wiring_private.h
*
************************************************************************ 
#include "wiring_private.h"
^
compilation terminated.
compilation terminated.
*** [.pioenvs\sonoff\src\core_esp8266_wiring_digital.c.o] Error 1
*** [.pioenvs\sonoff\src\core_esp8266_timer.c.o] Error 1
 [ERROR] Took 8.70 seconds
  [SUMMARY]
Environment sonoff-minimal      [SKIP]
Environment sonoff-classic      [SKIP]
Environment sonoff-knx          [SKIP]
Environment sonoff-allsensors   [SKIP]
Environment sonoff-BG           [SKIP]
Environment sonoff-BR           [SKIP]
Environment sonoff              [ERROR]

Sorry, guys - I'm at wits end! What the h**l am I doing wrong? This should be - **and has been**! - so easy and seamless. Now, it just appears nothing works! I've tried uninstalling and reinstalling Atom and PlatformIO on first computer - still just the same. Thanks for _any_ help!

Mikeinnc, 2018-07-14:
OK, well I've managed to successfully flash my two Sonoff basic devices using the Arduino IDE and latest Tasmota. Still not an easy option - a lot of changing files and updating versions and reading between the lines in a lot of confusing articles on-line - but, hey, what's new there!! So, nothing wrong with the boards or my connections etc - but still an error in PlatformIO. Why? I'd still like to get it working, as it certainly seems the superior method! Thanks....