arendst/Tasmota #3214: TM1638  Status from Switch are inverted in Message Sensor and Status 10

Jason2866, 2018-07-13:
```
00:00:00 Project sonoff WeMos159 (Topic sonoff-1A2AFC, Fallback sonoff-1A2AFC, GroupTopic sonoffs) Version 6.1.0a-2_3_0
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-1A2AFC-2812...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff-1A2AFC-2812 with IP address 192.168.2.159
00:00:06 UPP: Multicast (re)joined
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoff-1A2AFC/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoff-1A2AFC/POWER = 
00:00:07 MQT: tele/sonoff-1A2AFC/INFO1 = {"Module":"Generic","Version":"6.1.0a","FallbackTopic":"sonoff-1A2AFC","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff-1A2AFC/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1A2AFC-2812","IPAddress":"192.168.2.159"}
00:00:07 MQT: tele/sonoff-1A2AFC/INFO3 = {"RestartReason":"External System"}
00:00:08 UPP: Multicast (re)joined
18:31:04 MQT: tele/sonoff-1A2AFC/STATE = {"Time":"2018-07-13T18:31:04","Uptime":"0T00:00:15","Vcc":2.770,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":92,"APMac":"00:A0:57:2A:BD:19"}}
18:31:04 MQT: tele/sonoff-1A2AFC/SENSOR = {"Time":"2018-07-13T18:31:04","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"ON"}
18:32:33 CMD: Status 10
18:32:33 MQT: stat/sonoff-1A2AFC/STATUS10 = {"StatusSNS":{"Time":"2018-07-13T18:32:33","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"ON"}}
```

All switches are off

arendst, 2018-07-14:
There are two switch states:
- Normally the switch is not pressed and, in the case of the TM1638 this means the virtual switch is HIGH.
- When the switch is pressed it will set the virtual switch to LOW.

HIGH is not equivalent to ON and LOW is not equivalent to OFF. The relations are configured using command ``switchmode`` where options 2,4 and 6 will invert their states.

Think of it as a physical switch connected to a pulled-up GPIO. If the switch is not closed (= key on TM1638 is not pressed) the signal is HIGH and report as ON unless switchmode is set 2,4 or 6 when it will invert the states.

Sounds difficult but there was a reason why I did it that way suggested by users.

Jason2866, 2018-07-14:
Ok i understand, it is a design decision. Just need to know.
I will change -with switchmode- the behaviour to (for me! more logical) key press is ON

closed 2018-07-14 by Jason2866